devbisme/skidl #11: What is KISYSMOD environment variable

mrm2m, 2017-02-09:
I found it mentioned here, but I have no clue what it's supposed to do and what to with it. I can see that you use it to find your libraries. But instead of using the given path you use something like `/given/path/../library`. I have installed kicad on my linux system and my libraries are located in `~/projects/kicad/lib`. I think it's not a good idea to rely on the presence of this environment variable and then rely on the hard coded fact that libs are always in a directory called `library`.
If the environment variable is set up you may use it that way, but if it isn't there it should not trigger a warning (or make the program crash as shown in https://github.com/xesscorp/skidl/pull/10). 
There should be an alternative way to present more paths to search for libraries. If you use KISYSMOD it should be well documented.

G42makes, 2017-02-23:
I found I had to set it with an Ubuntu install of KiCAD. 
I generally just run my code with it on the command line:

```KISYSMOD=/usr/share/kicad/modules python 4keyboard.py```

I use that for interactive mode as well(without the .py file), when searching parts and looking at pinouts.

AlexSneedMiller, 2017-03-31:
I'm running into a similar issue on OSx. I have the [kicad](http://kicad-pcb.org/) installed, and the environment variable is set to `/Library/Application Support/kicad/modules/`. 

When I do something simple like `search('opamp')` it gives me an error saying that KISYSMOD is not define. When I manually set the environment variable in the command line then it says it can't find device.lib in that directory. Does anyone know where device.lib should be? The only file I have in there are a couple of the default files that come with kicad, such as "fp-lib-table".

xesscorp, 2017-04-01:
The way SKiDL currently works is it gets the path in the `KISYSMOD` variable, looks in the parent directory of that, and then looks in the child directory called `library` (so `library` and the directory pointed to by `KISYSMOD` are siblings). You need to look into that directory to see if the `.lib` files are in there. If not, you may have to manually install them from the [Github KiCad library repo](https://github.com/KiCad/kicad-library).

closed 2021-03-07 by xesscorp