devbisme/skidl #12: Subclassing 'Part' causes netlist issues

slightlynybbled, 2017-02-16:
I realize that the TEMPLATE style is intended for this, but I believe that I want to add methods to certain part types.

The basic voltage-divider sample, as copied from the documentation:

```
from skidl import *

gnd = Net('GND') # Ground reference
vin = Net('VI') # Input voltage to the divider
vout = Net('VO') # Output voltage from the divider

r1, r2 = 2 * Part('device', 'R', TEMPLATE) # Create two resistors
r1.value, r1.footprint = '1K', 'Resistors_SMD:R_0805' # Set resistor values
r2.value, r2.footprint = '500', 'Resistors_SMD:R_0805' # and footprints

r1[1] += vin # Connect the input to the first resistor
r2[2] += gnd # Connect the second resistor to ground
vout += r1[2], r2[1] # Output comes from the connection of the two resistors

generate_netlist()
```

I am attempting to perform a simple subclass of `Part` that is specifically for resistors so that I can easily add methods specific to resistors.  The same circuit:

```
from skidl import *

class Resistor(Part):
    def __init__(self, value, ref=None, footprint='Resistors_SMD:R_0805'):
        super().__init__('device', 'R', value=value, ref=ref, footprint=footprint)

gnd = Net('GND') # Ground reference.
vin = Net('VI')  # Input voltage to the divider.
vout = Net('VO')  # Output voltage from the divider.

r1 = Resistor('1k')
r2 = Resistor('500')

r1[1] += vin  # Connect the input to the first resistor.
r2[2] += gnd  # Connect the second resistor to ground.

vout += r1[2], r2[1]  # Output comes from the connection of the two resistors.

ERC()
generate_netlist()
```

In the attempt, I receive an error during netlisting.  Up to the `ERC()`, all works great.  The error:

```
ERROR: Can't generate netlist in an unknown ECAD tool format (<built-in function format>).
Traceback (most recent call last):
  File "C:\_virtualenv\skidl\lib\site-packages\skidl\skidl.py", line 1776, in _generate_netlist_component
    tool)]
KeyError: '_gen_netlist_comp_kicad'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
....
.... (more traceback)
```

Skidl is apparently seeting the ECAD tool format as incorrect when I subclass the Part.  I suspect that there is a simple fix, but I am unsure as to how to proceed.

Thanks,

j

xesscorp, 2017-02-16:
Thanks for the clearly-explained problem. I've got a fix for it. I'll try to get it out tonight.

closed 2017-02-17 by xesscorp

slightlynybbled, 2017-02-17:
Works perfectly.  Thank you for the excellent response time!

I heard your package on [podcast.__init__()](https://www.podcastinit.com/) and wanted to try it out.  Thank you for your work!  Finally, diffs on schematics!