devbisme/skidl #19: Factor code that parses and writes kicad file formats into a separate library

dvc94ch, 2017-04-12:
I think this is a generally useful component that many kicad scripting tools could use, no point in every tool rewriting or copy pasting this stuff.

xesscorp, 2017-04-13:
When parsing a KiCad netlist or symbol library, the file is processed and turned into Part and Net SKiDL objects. And when a KiCad netlist is output, that's done by calling the methods in the Part and Net objects that generate KiCad-specific statements. So there isn't a self-contained subroutine that does KiCad file I/O independent of the SKiDL objects. And if you need the SKiDL objects, then you effectively need the entire SKiDL library.

dvc94ch, 2017-04-15:
In case you are interested in collaborating on this, I added support for parsing kicad_mod files to pykicad. I'm also planning on adding support for .net, .kicad_pcb and .lib file formats.

[0] https://github.com/dvc94ch/pykicad

xesscorp, 2017-04-15:
Your pykicad looks like a useful project. As I said, I don't think my ad-hoc KiCad .lib parsing stuff is a good candidate. But I also have a [.net parser](https://github.com/xesscorp/skidl/blob/master/skidl/parse_netlist.py) that uses pyparsing, just like your module parser. It's relatively self-contained and you might find it's a useful starting point. However, there are two caveats:

1. It's slow, either because pyparsing is slow or because I'm doing it wrong.
2. My .net parser function isn't well documented. But at least it's small.

Another candidate for parsing is [PLY (Python Lex-Yacc)](https://github.com/dabeaz/ply). I don't know if it's faster than pyparsing or not.

dvc94ch, 2017-04-21:
I'll close the issue for now... pykicad features a schema based parser, so describing the kicad file formats should be trivial but a lot of work, with an extensive test suite. Let me know what if any features you'd like adding.

closed 2017-04-21 by dvc94ch