devbisme/skidl #21: Search with tool=SKIDL results in decode error

kasbah, 2017-07-21:
Works OK wihout `tool=SKIDL` (though it seems quite slow). 

```
In [1]: from skidl import *

In [2]: search('resistor', tool=SKIDL)
                                         ERROR: 'utf8' codec can't decode byte 0xc2 in position 236: invalid continuation byte
---------------------------------------------------------------------------     
Exception                                 Traceback (most recent call last)
<ipython-input-4-6ec6c2287cd2> in <module>()
----> 1 search('resistor', tool=SKIDL)

./skidl/skidl.pyc in search(term, tool)
   3996 
   3997     term = '.*' + term + '.*' # Use the given term as a substring.
-> 3998     parts = search_libraries(term, tool)  # Search for parts with that substring.
   3999 
   4000     # Print each part name sorted by the library where it was found.

./skidl/skidl.pyc in search_libraries(term, tool)
   3972             for lib_file in lib_files:
   3973                 print(' '*79, '\rSearching {} ...'.format(lib_file), end = '\r')
-> 3974                 lib = SchLib(os.path.join(lib_dir,lib_file), tool=tool) # Open the library file.
   3975 
   3976                 def mk_list(l):

./skidl/skidl.pyc in __init__(self, filename, tool, **attribs)
    650                 # Use the tool name to find the function for loading the library.
    651                 load_func = getattr(self, '_load_sch_lib_{}'.format(tool))
--> 652                 load_func(filename, lib_search_paths[tool])
    653                 self.filename = filename
    654                 # Cache a reference to the library.

./skidl/skidl.pyc in _load_sch_lib_skidl(self, filename, lib_search_paths)
    781         except Exception as e:
    782             logger.error(e)
--> 783             raise Exception
    784 
    785     def add_parts(self, *parts):
```

xesscorp, 2017-07-22:
I tried this with Python 3.5.2 and 2.7.12 and a pip install from pypi of the latest SKiDL release. It works with no errors.
Let me know your setup so I can try to recreate the error.

kasbah, 2017-07-22:
This is on Ubuntu 14.04. It happens with Python 2.7.6, 3.4.3 and 3.5.3. I had a daily build of KiCad installed but switching back to stable made no difference either.

kasbah, 2017-07-22:
I put in a print to get the filename of the file that's causing trouble and it's `conn_sklib.py`. All other libs are either UTF-8 or ASCII but that one. If I remove it everything is fine. 

```
conn_sklib.py:                 Python script, ISO-8859 text executable, with very long lines
contrib_sklib.py:              Python script, ASCII text executable
convert_libs.py:               Python script, ASCII text executable
cypress_sklib.py:              Python script, UTF-8 Unicode text executable, with very long lines
```

PR #22 (2017-07-24): Change conn_sklib encoding to UTF-8
fixes #21
commit 0162e01c (2017-07-24): Change conn_sklib encoding to UTF-8

fixes #21
closed 2017-07-24 by xesscorp