devbisme/skidl #25: RFC: easy_skidl - parsing component descriptions

kasbah, 2017-10-17:
I made a component description parser called [Electro Grammar](https://monostable.github.io/electro-grammar/) that currently works on SMD resistor, capacitors and LEDs (planning to expand this to other types of components as well). 

I have been porting it to Python (still very much work-in-progress) and thought I would try and use it to make SKiDL a bit easier. Here is an example of what I have been playing around with today: 

```python
import skidl
from easy_skidl import *

r1 = resistor('1k 0603')
r2 = r1.copy()

c1 = capacitor('0.1uF 0805')

led1 = led('0805 red')

vin = skidl.Net('VIN')
gnd = skidl.Net('GND')

vin     += r1[1], r2[1]
r2[2]   += c1[1]
c1[2]   += gnd
led1[1] += r1[2]
led1[2] += gnd
```

[more details in a notebook](https://nbviewer.jupyter.org/gist/kasbah/1ab5e25ab27805b40fba8b69397fa4fc)

Some things to note:

- those `easy_skidl` function currently break the `*` comprehensions i.e. `r1, r2 = resistor('1k') * 2` won't work, do you see a good way around that?
- I am planning to license Electro Grammar as AGPLv3 (is this an issue?)

Any comments on this at all are very much appreciated.

xesscorp, 2017-10-18:
This looks good to me. What are the implications of using nearly.js? Does this drag in some Javascript dependencies.

I noticed you're using quantiphy. How does this compare to using Pint? (At least on Github, Pint has much more favoriting/forking.)

I don't concern myself too much with licensing, as long as people are free to use stuff and maintain as much control as possible over what they create. Keeping easy_skidl and Electro Grammar as separate entities allows people to choose whether to use them or not.

kasbah, 2017-10-18:
>This looks good to me. What are the implications of using nearly.js? Does this drag in some Javascript dependencies.

No, the python version of Electro Grammar is transformed using [Lark](https://github.com/erezsh/lark) and does not have any JS dependencies after transformation. 

>I noticed you're using quantiphy. How does this compare to using Pint? (At least on Github, Pint has much more favoriting/forking.)

No idea, It was just the first thing I found when I realised that my values where in floating point and that SKiDL expects strings. 

Re: licensing, sounds sensible. 

One thing I had been thinking of is allowing `Part` to be instantiated with a description instead of using a separate function. So being able to do:

```
from skidl import Part
c1 = Part('1uF 0603')
```

What do you think?