devbisme/skidl #29: Bit confused about `Bus`

kasbah, 2017-10-18:
I have some questions regarding buses. I feel I understand buses in schematics but maybe I haven't used them very much. In the context of SKiDL I don't understand:

- What does a bus provide as opposed to a list of nets?
- Why do you specify the width of a bus if it's dependent on the number of nets in it?
- Why do buses need to be named?

xesscorp, 2017-10-20:
* Buses provide a way to extract portions of the bus using things like name strings. And the individual nets will have the base name of the bus. But for many operations, you could just use a Python list containing Net objects you've created.

* You don't have to specify a bus width. You can build a bus from any combination of widths, existing nets, or existing buses and the resulting bus width will be the sum of the widths of its members.

* Buses probably shouldn't need to be named. They could have randomly assigned names like Net objects. Usually you wouldn't want that: you would like to see a recognizable bus name listed in your PCB layout. But it probably should be allowed to have unnamed buses.

kasbah, 2017-10-20:
Thanks for explaining. I guess the only thing I am still confused about is: why would you want to specify a bus width in advance when it will dynamically change when adding nets to it?

xesscorp, 2017-10-25:
You can define a bus as a combination of widths, nets, and buses. For example, this is a bus made up of eight bus lines, two existing nets, and another bus:

b = Bus('new_bus', 8, net1, net_b, other_bus)

Bus b now has a width of 8 + 2 + len(other_bus). You can make the bus wider using extend() or insert() like so:

b.extend(another_bus)

Now the bus has a width of 8 + 2 + len(other_bus) + len(another_bus). There is no way to make the bus narrower once you've added lines to it. Although you could add more lines to the bus at any time, I imagine most of the time the bus will be defined once with a given width.

Once the bus is defined, then you can connect it to pins or other nets and buses using the += operator. This does not change the width of the bus; it just makes electrical connections between the bus and other things.

kasbah, 2017-11-17:
>There is no way to make the bus narrower once you've added lines to it. Although you could add more lines to the bus at any time, I imagine most of the time the bus will be defined once with a given width.

I guess to me it is an odd choice to have it behave this way. As far as I can tell there is no reason to define the width of bus up front. Maybe to understand it I have to actually try and use it.

closed 2017-11-26 by kasbah