devbisme/skidl #37: visualizer/viewer

nkkPb, 2018-03-08:
It would be awesome if you could build a graphical schematic from the netlist file (or python script). you could integrate it to letex for viewing or have the flexibility of switching from and to a graphical schematic digram.

kasbah, 2018-03-08:
Hey nkkPb, I added basic graphviz support. Check the `generate_graph` method. [Here are some examples](https://nbviewer.jupyter.org/gist/kasbah/cedfcb65327c7efe1565eb68fed1c5b0). 

For [Pycircuit](https://github.com/dvc94ch/pycircuit) and [Replicad](https://github.com/monostable/replicad) we want to enable visualization through [netlistsvg](https://github.com/nturley/netlistsvg). Pycircuit has some support already but it's still experimental I believe. 

For SkiDL to join the party the first step would be to enable Yosys netlist output. There's still a lot to figure out in Netlistsvg though especially to do with rotation and junction nodes though.

nkkPb, 2018-03-08:
awesome. thanks for the info.

nkkPb, 2018-03-08:
what is the current progress on generating netlist formatted for Yosys?

kasbah, 2018-03-08:
Nothing of the sorts in the pipeline yet as far as I know. I am pretty sure @xesscorp wouldn't object to an additional netlist output format though.

psychemedia, 2019-08-13:
For reference, `lcapy` has support for generating schematics from its internal representation. Some examples [here](https://nbviewer.jupyter.org/github/psychemedia/showntell/blob/electronics/3.6.0%20Electronics.ipynb)

closed 2021-07-09 by xesscorp