devbisme/skidl #40: Adding placement and/or routing functionality

NikiSchlifke, 2018-11-21:
I would like to do some brainstorming about the possibility of adding placement and / or routing hints.
# Problem
After generating a netlist and importing it into pcbnew you are left with the still tedious task of placing the footprints, drawing the traces and general layout.
But I think functionality can be added to include at least placement hints along with the generated netlist as in many cases you already know in loose terms where a particular part should be.
# Possible approaches
## Include information about part placement in the netlist
For example: (place parts right edge at 20 mil from the origin, center 100 mil from the origin)
``` JSON
{"right": "20mil", "y": "100mil"}
```
Maybe just add it into the value field, then use a python script from pcbnew to parse the placement data if it contains json with the relevant fields.
## Generate kicad_pcb file from the start
Since the file formats of .net and .kicad_pcb are fairly similar it might be more convenient to go ahead and generate a kicad_pcb file that already has the parts placed.
Any tips where to start in regards to existing skidl code?
It might be useful to come up with an intelligent approach where an existing file isn't overwritten but "patched" with the new position.

xesscorp, 2018-11-26:
I'm just back from vacation. I also would like to see some features like this added to SKiDL. Let me think about it and come back later.

xesscorp, 2018-12-04:
I think you can add some placement information to a Part object directly as you showed above. Part rotation would be needed as well. We would also need to be able to place a block of parts and have all the parts included in that block placed accordingly. So that will require some geometric transformation routines applied as you descend a hierarchy. Parts might also be placed relative to other parts in a block (like "part U1 to the right of U2").

I would like SKiDL to output some kind of part placement file in addition to the netlist file. Then a separate PCBNEW plugin could read the placement file and move the components. This plugin would be useful with other utilities that might generate placement files, not just SKiDL.

I'm not really for SKiDL generating PCBNEW files directly. Just generate the netlist and placement file and import those into PCBNEW and you're done.

cejpmart, 2019-01-26:
Just to clear this up; even if I know the exact placement of all parts, at the moment there is no way to relay this information through SKiDL to Pcbnew?

xesscorp, 2019-01-26:
No, SKiDL does not currently do anything with part placement. It's not conceptually hard to do, but it would have to output the part placements to another file as the netlist file doesn't have a place for that information. Then you'll also need a plugin for PCBNEW to parse that part position file and move the components.

andrewkenny, 2019-03-22:
+1 on this,
I am also interested in placing components (and in the future tracks) using python or similar.

It appears that the simplest format would be a csv file.

single pin devices such as vias or testpads or an antenna connection are presumably simple:

> Device, X, Y, angle
where:
Device - Device name (dur)
X, Y     - where to place these components WRT tooling origin / fiducial. an X, Y of 0, 10 would mean that the center of the component would be 10mm from the tooling origin
Angle  - how much to rotate the device, where 0 degrees / radians is horizontal (to match sin(0) = 0)

If the device package is given, then I would imagine it would be easy to calculate the location of the subsequent pins, should somebody in the future want to use this for some simple routing.

henyhollar, 2019-09-14:
I am also interested in this feature on SKiDL. 
I stumbled on an interesting project implementing something close to this:
https://xesscorp.github.io/kinjector/docs/_build/singlehtml/index.html

xesscorp, 2019-09-14:
Yes, I built KinJector specifically for injecting net classes and part positions from SKiDL code into the PCB. I've got most of the net class stuff done, but still haven't gotten to the part placement.