devbisme/skidl #44: skidl has trouble with the Digikey libraries due to encoding

AppMathIsRealMath, 2019-02-10:
I've installed the Digikey libraries from here on GitHub into Kicad. They aren't UTF-8 clean encoded and this gives skidl fits:

Python 3.6.8 (default, Jan  2 2019, 05:35:58) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from skidl import *
>>> search('opamp')
lib_dir = 
WARNING: Could not open directory ''
lib_dir = .
lib_dir = /mnt/tools/kicad/share/kicad/library
Traceback (most recent call last):h-Screen-Controllers.lib ...                  
  File "<stdin>", line 1, in <module>
  File "/home/sam/.python/3/lib/python3.6/site-packages/skidl/part_query.py", line 103, in search
    tool)  # Search for parts with that substring.
  File "/home/sam/.python/3/lib/python3.6/site-packages/skidl/part_query.py", line 75, in search_libraries
    tool=tool)  # Open the library file.
  File "/home/sam/.python/3/lib/python3.6/site-packages/skidl/SchLib.py", line 96, in __init__
    load_func(filename, skidl.lib_search_paths[tool])
  File "/home/sam/.python/3/lib/python3.6/site-packages/skidl/tools/kicad.py", line 121, in _load_sch_lib_
    for line in f:
  File "/home/sam/.python/3/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x99 in position 102: invalid start byte

When I use iconv to force encode them to UTF-8 skidl is happy.

Great work on skidl and your other tools! I share your aversion to schematic capture.

xesscorp, 2019-02-10:
Thanks for the bug report! I'll look into the character encoding issue. 

BTW, if you have any disclosable designs using SKiDL, send me a link. I haven't seen many so it's hard to tell what features people like/dislike or use/avoid. So I have to rely on my own judgement and that's a recipe for disaster.

xesscorp, 2019-02-24:
I changed the encoding SKiDL uses to read files from utf8 to latin_1. That seems to fix the problems caused when strange symbols like Ohms are encountered.

closed 2019-02-24 by xesscorp