devbisme/skidl #46: Separate meta data from instance data

obiwanjacobi, 2019-05-05:
Disclaimer: Skidl and Python noob ;-)

I would like an (optional) alternate way to instantiate parts. The duplication in the Part's constructor can be prevented by first creating a meta definition. This also allows a central definition of common used parts.

=>part-like constructor of meta data
partInfo1 = PartInfo(....)
partInfo2 = PertInfo(....)

=> no need to think about the specific part.
part1a = Part(partInfo1)
part1b = Part(partInfo1)
part2a = Part(partInfo2)
part2b = Part(partInfo2)

xesscorp, 2019-05-05:
Let me offer you two other ways to do this and see if either of them meets your needs:

First, you could make a file of common parts containing things like this:

```py
r47k_t = Part('Device', 'R', value='47K', dest=TEMPLATE)
c1000nf_t = Part('Device', 'C', value='1uF', dest=TEMPLATE)
...
```

Then, you could instantiate parts like this:

```py
r1 = r47k_t()
c1 = c1000nf_t()
```

The second way is to use the `**` operator to expand dictionaries inside function calls. In this case, make a file containing common part information:

```py
r47k_pi = {'lib':'Device', 'name':'R', 'value':'47K'}
c1000nf_pi = {'lib':'Device', 'name':'C', 'value':'1uF'}
...
```

Then, use the part information dictionaries like this:

```py
r1 = Part(**r47k_pi)
c1 = Part(**c1000nf_pi)
```

Both these techniques do the same thing. The second technique seems closest to what you want with the difference that you'll need to use the ** operator.

obiwanjacobi, 2019-05-06:
I see I am missing some Python knowledge...
Thanks.

closed 2021-07-09 by xesscorp