devbisme/skidl #47: Footprints not found pcbnew import

obiwanjacobi, 2019-05-05:
Made a simple start of a clock schematic with Skidl and generated a netlist without errors.
When I try to import the netlist into pcbnew (kicad 5.1.0) it gives errors saying it cannot find the footprints.

To test this I dropped a 7400 onto a schematic and exported its netlist. The footprint name is exactly the same as I used in my skidl code, but pcbnew does not give any errors...?

So what am I doing wrong?

Here's the code:

```
from skidl import *

# libraries
logic74 = '74xx'
device = 'device'

# footprints
fpR = 'Resistor_SMD.pretty:R_0805_2012Metric'
fpC = 'Capacitor_SMD.pretty:C_0805_2012Metric'
fpXTAL = 'Crystal.pretty:Crystal_HC18-U_Vertical'
fpU14 = 'Package_SO.pretty:SOIC-14_3.9x8.7mm_P1.27mm'

# global nets
vcc = Net('VCC')
gnd = Net('GND')

# clock 1.19
# http://www.zxdesign.info/docs/clock-1.19.pdf

xtal = Part(device, 'crystal', footprint=fpXTAL)
r1 = Part(device, 'R', value='470R', footprint=fpR)
r2 = Part(device, 'R', value='470R', footprint=fpR)
c1 = Part(device, 'C', value='100pF', footprint=fpC)
u1 = Part(logic74, '74LS00', footprint=fpU14)

u1[7] += gnd
u1[14] += vcc
u1[12] += vcc
u1[9] += vcc

xtal[1] += r1[1]
xtal[1] += u1[13]

r1[2] += c1[1]
r1[2] += u1[11]

r2[2] += c1[2]
r2[2] += u1[10]

xtal[2] += r2[1]
xtal[2] += u1[4]
xtal[2] += u1[5]
xtal[2] += u1[8]

clk14 = Net('Clk14')
r54 = Part(device, 'R', value='4K7', footprint=fpR)
r54.ref = 'R54'

r54[1] += vcc
r54[2] += clk14
r54[2] += u1[6]
#--------------------------------
generate_netlist()
```

Here is the example of the generated netlist 7400 part:

```
(comp (ref U1)
      (value 74LS00)
      (footprint Package_SO.pretty:SOIC-14_3.9x8.7mm_P1.27mm)
      (fields
        (field (name keywords) "TTL nand 2-input buffer")
        (field (name description) "quad 2-input NAND buffer"))
      (libsource (lib 74xx) (part 74LS00))
      (sheetpath (names /top/17017331776065520654) (tstamps /top/17017331776065520654)))
```

And here is the 7400 part generated from the schematic

```
(comp (ref U1)
      (value 74HC00)
      (footprint Package_SO:SOIC-14_3.9x8.7mm_P1.27mm)
      (datasheet http://www.ti.com/lit/gpn/sn74hc00)
      (libsource (lib 74xx) (part 74HC00) (description "quad 2-input NAND gate"))
      (sheetpath (names /) (tstamps /))
      (tstamp 5CCF1A6A)))
```

I assume it should be possible to import the skidl generated netlist into pcbnew, yes?

xesscorp, 2019-05-05:
What version of KiCad are you using? That *may* have something to do with whether it likes the ".pretty" extension or not.

obiwanjacobi, 2019-05-06:
kicad 5.1.0

tailsu, 2020-12-27:
Confirmed. At least on version 5.1.8 the footprint string should not contain the `.pretty` extension.

closed 2021-07-09 by xesscorp

ibkvictor, 2022-11-15:
`Error: Cannot add R2 (footprint 'Resistor_SMD.pretty:R_0805_2012Metric' not found).` Having the same error on KiCad version 6.0.8. Please, what can I do?

devbisme, 2022-11-15:
Please provide the code that caused this error in addition to the version of SKiDL, python, and your OS. Thanks!

ibkvictor, 2022-11-16:
Thanks, I fixed it by removing `.pretty` in the footprint argument. I think it is an issue with kiCad version 6 using a new way of defining library footprints. Thanks once again. Is there a gitter or community platform for skidl. I would like to know how best to transpose my netlist in spice `.sp` or json `.json` to a kiCad schematic using skidl.