devbisme/skidl #48: Composability of subcircuits

flaviut, 2019-05-08:
Let's say that I have a subcircuit for the main part of a buck converter, and I've parameterized it to accept an arbitrary inductor, diode, etc:

```python

def buck_converter(
        nmos: Part,
        diode: Part,
        inductor: Part,
        cap_input: Part,
        cap_output: Part,
):
    @subcircuit
    def buck_impl(pwm: Net, v_in: Net, v_out: Net, gnd: Net):
        buck_inner_net = Net('buck_inner')

        nmos_inst = nmos.copy()
        nmos_inst['gate'] = pwm
        nmos_inst['drain'] = v_in
        nmos_inst['source'] = buck_inner_net

        inductor_inst = inductor.copy()
        inductor_inst['+'] = buck_inner_net
        inductor_inst['-'] = v_out

        diode_inst = diode.copy()
        diode_inst['-'] = buck_inner_net
        diode_inst['+'] = gnd

        cap_input_inst = cap_input.copy()
        cap_input_inst['+'] = v_in
        cap_input_inst['-'] = gnd
        cap_output_inst = cap_output.copy()
        cap_output_inst['+'] = v_in
        cap_output_inst['-'] = gnd

    return buck_impl
```

However, what if I want to replace the diode with synchronous rectification? What I'd like to be able to do is create another subcircuit that provides that functionality, and be able to pass it in as the `diode` parameter. As long as it has the same pin names, why shouldn't I be able to do that?

However, I can't do that, as the Circuit interface is nothing like the Part interface, and I can't create a part with multiple components.

It seems to me that idea of a Circuit isn't a very nice abstraction. It serves as a bag of parts and nets, a duplicate of the data that is already stored in the part and net graph. Why can't that bag of parts and nets be built up as it is needed?

flaviut, 2019-05-08:
I've re-read this, and I realize that my tone probably isn't coming across too great.

I want to emphasize that I've really enjoyed working with skidl in the brief amount of time that I have, and that it's the whole idea of circuits through code is really great.

xesscorp, 2019-05-08:
I like the idea of creating a Part-like interface around a Circuit. I'll have to think about how that might be most-easily done.

I also like how you structured your example to create a factory that returns a subcircuit that can be easily instantiated and connected without having to specify the parts from which it's constructed each time.

flaviut, 2019-05-09:
I have an idea on how to implement this, but it's basically refactoring the whole thing (in a non-backwards-compatible way :smile:):

Have only two main types:

```
Base (
    attrs: Dict[str, Any]
)

Net extends Base (
    name: str,
    connected_to: List[Union[Tuple[Part, Pin], Net]],
)

Part extends Base (
    name: str,
    pins: Dict[Pin, List[Net]],
)
```

We can use nothing but these types to implement skidl's features, in what I think is a cleaner, more extensible way. I'm going to go over the following items since I think they're representative of what skidl does:

- Subcircuits
- Multi-unit parts
- Part reference assignment

This is done by realizing that skidl is really the same thing as a compiler. It takes a graph input, applies some transformations to that graph, and produces an output.

There are some differences, like how [ASTs][] are directed acyclic graphs, while a circuit is an undirected cyclic graph. Another major difference is that skidl doesn't have to deal with parsing source code into a graph, instead the user generates the graph directly.

[asts]: https://en.wikipedia.org/wiki/Abstract_syntax_tree

## Subcircuits

A subcircuit is just a Part that has no footprint. It has pins, and when it is created, the subcircuit's pins are assigned to the nets & parts that form the implementation of the subcircuit.

```python
def voltage_divider(…) -> Part:
    voltage_divider = Part(pins=['vin', 'out', 'gnd'])
    vin_net = Net()
    out_net = Net()
    gnd_net = Net()

    r1 = Part(…'R'…)
    r2 = Part(…'R'…)
    r1[0] += vin_net
    r1[1] += out_net
    r2[0] += out_net
    r2[1] += gnd_net

    return voltage_divider
```

The above code is quite verbose, but perhaps with some syntax sugar it could be written as:

```python
def voltage_divider(…) -> Part:
    voltage_divider = Part(pins=['vin', 'out', 'gnd'])

    r1 = Part(…'R'…)
    r2 = Part(…'R'…)
    r1[0] += voltage_divider['vin']
    r1[1] += voltage_divider['out']
    r2[0] += voltage_divider['out']
    r2[1] += voltage_divider['gnd']

    return voltage_divider
```

I'll be using this syntax from now on since I hate typing.

## Multi-unit parts

This is just a certain kind of subcircuit, where multiple parts are created that correspond to the different pins of a single part

```python
def opamp_subparts(input_amp, …) -> Tuple[Part, Part, Part]:
    """Splits a 2-channel opamp into parts"""
    power = Part(pins=['vin', 'gnd'])
    op_1, op_2 = [Part(pins=['+', '-', 'out'])] * 2

    input_amp['vin'] += power['vin']
    input_amp['gnd'] += power['gnd']

    input_amp['+_1'] += op_1['+']
    input_amp['-_1'] -= op_1['-']
    input_amp['out_1'] -= op_1['out']

    input_amp['+_2'] += op_2['+']
    input_amp['-_2'] -= op_2['-']
    input_amp['out_2'] -= op_2['out']

    return (power, op_1, op_2)
```

## Part reference assignment

This is the fun part, where we can take some inspiration from compiler theory. Instead of constantly trying to avoid introducing duplicates while constructing the circuit (and using a global variable to take care of that), we instead transform our graph before sending it to the netlist generator:

```python
def iterate_parts(input: List[Union[Net, Part]]) -> Generator[Part, None, List[Union[Net, Part]]]:
    # deeply copies the graph and yields each part in the graph
    # the deep copy is important because
    #
    # 1. copying it avoids different processing passes stepping on each others toes
    # 2. makes it easier to transform the graph since passes can mutate the objects

def assign_ids(input: List[Union[Net, Part]]) -> List[Union[Net, Part]]:
    last_ids = defaultdict(default=1)
    try:
        for part in iterate_parts(input):
            if part.attrs['footprint'] is None:
                continue

            id_prefix = part.attrs['id_prefix']
            part.attrs['id'] = id_prefix + last_ids[id_prefix]
            last_ids[id_prefix] += 1
    except StopIteration as ex:
        return ex.value
```

The important thing to note here is that the full responsibility for generating the IDs is in the `assign_ids`. There's no additional code needed in the Part class.

The same process can be repeated for the other functionally, like generating XML. Instead of having special methods on the Part class, it could be done in a separate method, in a separate file, or even another person's code.

---------

I'd love to hear your thoughts on this.

xesscorp, 2019-05-09:
The Circuit object didn't come along until the latter part of SKiDL development. It was kind of a grab-bag of pieces. Re-thinking it as a type of Part makes sense.

I'm concerned about your mention of "non-backwards compatible". To me, that means there would be current features of SKiDL that would not be available in a future version that used your ideas. I can't really see why that would be. Maybe you mean something else.

I think getting the composability feature you want can be done with some tweaking of the Circuit object in the current version of SKiDL. However, a refactoring may lead to simplification and clearing of some of SKiDL's underbrush. To that end, maybe I could make a branch where you could flesh-out some of your ideas and see where it leads.

flaviut, 2019-05-09:
> I'm concerned about your mention of "non-backwards compatible". To me, that means there would be current features of SKiDL that would not be available in a future version that used your ideas. I can't really see why that would be. Maybe you mean something else.

I mean that if these changes were to be made, then existing code would break. This wouldn't be pleasant for users, as they would have to migrate their code over or stick with an old version. 

All existing functionality would still exist, but it might (or might not) be accessed in a different way.

> However, a refactoring may lead to simplification and clearing of some of SKiDL's underbrush. To that end, maybe I could make a branch where you could flesh-out some of your ideas and see where it leads.

Yup, I'll go ahead and flesh it out in a branch. I just wanted to see if you think this is a good idea, since you're much more familiar with the code than I am.

I already [have a fork of this repo](https://github.com/flaviut/skidl), I'll send in a PR when I have something that more or less works so that I can get some feedback.

matthuszagh, 2020-01-15:
Hi @flaviut, I'm curious if you've made any progress with this, as I'm also very interested in this functionality/think it's a great idea.

xesscorp, 2020-05-27:
I [posted](https://xesscorp.github.io/skidl/docs/_site/blog/package-decorator) about SKiDL's new `@package` decorator that I hope solves some of this issue. Take a look and let me know your thoughts.

flaviut, 2020-05-27:
@matthuszagh I'm sorry to say I haven't made any real progress with this :(

But I'm really liking the `@package` decorator from above! Seems to do exactly this, without breaking backwards-compatibility. Thank you, and nice job!

closed 2020-05-27 by flaviut