devbisme/skidl #52: Units: Cannot use the same alias for multiple pins

obiwanjacobi, 2019-05-15:
Now this may be me not knowing python :-)

This code
```
part = Part('74xx', '74LS32')
part.make_unit('A', 1, 2, 3)
part.A.set_pin_alias('1A', 1)
```
gives this error
> ERROR: Cannot use the same alias for multiple pins.

Looking at the Part.py source I think somehow my pin '1' is interpreted as a list of pins.
Not sure why.

Also I was surprised that the units were not already there, I can see them in the KiCad symbol editor? I call set_default_tool(KICAD) to make sure those are used....?

xesscorp, 2019-05-21:
Apologies for the delay. Medical and computer issues...

The cause of the error is that the 74LS32 is defined in that library with multiple pins having the same pin number. So, for example, pin 1 is listed twice in the symbol definition: once as a normal input pin, and a second time as an *invisible* input pin. (I don't know why they do that. I'll have to investigate further.) So SKiDL sees two 1 pins and assumes you're assigning the same alias to both of them and throws an error.

The reason the units are not there is that SKiDL doesn't create them automatically when the symbol is read from the library file. This is fixed in the next release.

xesscorp, 2019-05-21:
OK, I figured out why there are duplicated pins: some symbols in the 74XX library contain both the "normal" logic gate and the *DeMorgan* equivalent. So I'm going to have to modify the KiCad library reader to ignore those in the next release.

obiwanjacobi, 2019-05-22:
Is there an ETA on that next release?

xesscorp, 2019-05-22:
I'm getting ready to push it to Github. Once I do that, I'll let you know how to install it. (It's not ready for a full PyPi release.) This push will also enable automatic loading of part units when the symbol library is read, and it makes creating pin aliases a bit easier.

closed 2019-05-22 by xesscorp

xesscorp, 2019-05-22:
The newest modifications have been pushed. You can get them like this:

    pip install -U https://github.com/xesscorp/skidl/archive/master.zip

Now when a part is read from a symbol library, the part units are automatically created with names like uA, uB, ...

```py
Python 2.7.16 (v2.7.16:413a49145e, Mar  4 2019, 01:30:55) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from skidl import *
>>> part = Part('74XX', '74LS32')
WARNING: Non-identical pins with the same number (1) in symbol drawing 74LS78
WARNING: Non-identical pins with the same number (5) in symbol drawing 74LS78
>>> part.unit    <= Print all the part units.
{u'uE':
 74LS32 (): Quad 2-input OR
    Pin U1/14/VCC/POWER-IN
    Pin U1/7/GND/POWER-IN, u'uD':
 74LS32 (): Quad 2-input OR
    Pin U1/13/~/INPUT
    Pin U1/11/~/OUTPUT
    Pin U1/12/~/INPUT, u'uA':
 74LS32 (): Quad 2-input OR
    Pin U1/2/~/INPUT
    Pin U1/1/~/INPUT
    Pin U1/3/~/OUTPUT, u'uC':
 74LS32 (): Quad 2-input OR
    Pin U1/9/~/INPUT
    Pin U1/8/~/OUTPUT
    Pin U1/10/~/INPUT, u'uB':
 74LS32 (): Quad 2-input OR
    Pin U1/6/~/OUTPUT
    Pin U1/5/~/INPUT
    Pin U1/4/~/INPUT}
```

For making pin aliases, just make an assignment to the .aliases member of a part or part unit. You can also assign multiple aliases to a pin by using the += operator.

```py
>>> part.uA[1].aliases = '1A'
>>> part.uA['1A']
Pin U1/1/~,1A/INPUT     <= Pin 1 now has alias 1A.
>>> part.uA[1].aliases += 'A1'
>>> part.uA[1]
Pin U1/1/~,A1,1A/INPUT    <= Pin 1 now has aliases 1A and A1.
```

commit 97d92d55 (2019-05-22): Fix #52 caused by multiple pins with the same number due to DeMorgan equivalents in the symbol.
* Fix TypeError caused when an unmatched pin number falls through to a block where it is used as a string.
* Reduce tox testing to only Python 2.7 and 3.7.