devbisme/skidl #53: Improving performance

aselle, 2019-05-17:
I love skidl, it is just what I need for reducing tediousness of digital designs. However, I find performance to be slow. If I have a python script that generatets a netlist, it takes about 15 seconds for it to load libraries and get through the skidl builders. I feel like it likely the parse time of libraries, but the profiling output wasn't super obvious. 

Does anyone have any workaround for this issue (caching, etc). I find I want to iterate quickly as I incrementally build up a circuit. Maybe some caching would help?  I'm going to play with mini_reset(), but I tend to have a workflow where I reload() a class that contains a circuit so I might lose the instance in my current scheme.

Thanks for the great software!

aselle, 2019-05-17:
So the `mini_reset` seems to do the trick... it's a bit awkward
if I define a class in foo.py:
```python 
class Foo:
	def __init__(self, circuit):
		self.circuit = circuit
		self.circuit.mini_reset()
		self.build(circuit=self.circuit)

	@subcircuit
	def build(self):
		self.gnd = Net("GND")
		self.vcc = Net("VCC")
		vo = Net("VO")
		p1 = Part("device", "R", value="1k")
		p2 = Part("device", "R", value="1k")
		self.vcc += p1[1]
		vo += p1[2]
		vo += p2[1]
		self.gnd += p2[2]
```		
I can then do 
```
python
>>> circuit = Circuit()
>>> reload(foo);a = foo.Foo(circuit)
```

pepijndevos, 2019-07-12:
I'm trying to use Skidl on a netlist of about 1400 chips, it takes a few minutes to build the circuit, and... it has not yet completed netlist generation. I tried a smaller one of about 500 and that eventually finished. I also tried PyPy, which is for some reason even slower. Have not yet done any real profiling yet.

xesscorp, 2019-07-12:
If you can share your circuit, I'd love to do some profiling on it. If not, can you give me some stats on how bigh these chips are and how they are connected? I might be able to build some similar random circuits and run some benchmarks.

pepijndevos, 2019-07-12:
There you go:
[benchmark.zip](https://github.com/xesscorp/skidl/files/3386815/benchmark.zip)
Alternatively (requires Yosys to generate the JSON file):
https://github.com/pepijndevos/74xx-liberty/blob/kicad/kicad/generate_netlist.py

xesscorp, 2019-07-15:
Thanks for the benchmark. It's a great test. The master branch will compile the netlist in 2160 seconds. The speedup branch reduces this to 350 seconds. Not great, but better than what it was. It will require a bit more surgery to get it to go faster.

pepijndevos, 2019-07-15:
Awesome. Great work.

xesscorp, 2019-07-22:
I've reduced the running time for compiling the picorv32 from 2160 down to 230 seconds. I've merged the speed-up changes into the master branch and removed the speedup branch. That's all I plan to do right now, although there is much room for improvement.

closed 2019-12-18 by xesscorp

aselle, 2019-12-18:
Thanks for all your work on Skidl! This is a great improvement.

xesscorp, 2019-12-18:
You're welcome!