devbisme/skidl #54: UnicodeDecodeError when generating netlist

RCoeurjoly, 2019-05-31:
I am just trying to replicate the generation of the Arduino netlist using skidl.

I am using the last version of both KiCAD and skidl and I get this error when executing the skidl script.

> Traceback (most recent call last):
  File "arduino_skidl.py", line 678, in <module>
    generate_netlist()
  File "/usr/local/lib/python2.7/dist-packages/skidl/Circuit.py", line 515, in generate_netlist
    netlist = gen_func(**kwargs)  # Pass any remaining arguments.
  File "/usr/local/lib/python2.7/dist-packages/skidl/tools/kicad.py", line 458, in _gen_netlist_
    netlist += '\n' + p.generate_netlist_component(KICAD)
  File "/usr/local/lib/python2.7/dist-packages/skidl/Part.py", line 698, in generate_netlist_component
    return gen_func()
  File "/usr/local/lib/python2.7/dist-packages/skidl/tools/kicad.py", line 500, in _gen_netlist_comp_
    fld_value = add_quotes(self.__dict__[fld_name])
  File "/usr/local/lib/python2.7/dist-packages/skidl/utilities.py", line 278, in add_quotes
    return '"' + s + '"'
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 26: ordinal not in range(128)

xesscorp, 2019-05-31:
I always have problems with this. Some non-ASCII characters were used in the part descriptions for "micron" and "Ohms" and such. I thought I fixed some of those bugs, but maybe that was just when generating the SKiDL from the netlist, and not when actually executing the SKiDL code.

I recommend re-trying this using the latest development code for SKiDL:

    $ pip install -U git+https://github.com/xesscorp/skidl

If that doesn't solve the problem, then send me the generated `arduino_skidl.py` file so I can look at it.

RCoeurjoly, 2019-05-31:
I have regenerated the netlist from KiCAD, executed the netlist_to_skidl utility, and executed the resulting SKiDL code, and the results are exactly the same.

You can find the .py in [this directory](https://github.com/RCoeurjoly/Arduino-in-SKiDL).

Couldn't we just use UTF-8 everywhere? What is stopping us from doing so?

xesscorp, 2019-06-02:
Thanks for the file. I was able to fix the problem. Go ahead and update your SKiDL from the repo like before and see if it works for you.

I'd like to use UTF-8 everywhere, but there seem to be a lot of places to fix when running under Python 2.7. (The existing code worked fine for me under Python 3.) Since Python 2 will lose support at the end of this year, I'm not highly motivated to fix this problem. If you know an easy way to fix it, then let me know or send a PR.

RCoeurjoly, 2019-06-02:
Thanks. I have updated SKiDL and this problems disappears.
I have another problem that I will open in another Issue.

closed 2019-06-02 by RCoeurjoly