devbisme/skidl #58: Unexpectedly long results for search

RCoeurjoly, 2019-06-13:
Upon searching 'opamp', I get the following results, with a lot of warnings from unrelated parts.
What is this due to?
Can someone replicate this issue?
Also, it takes a lot of time to execute the search.

> python 
Python 2.7.12 (default, Nov 12 2018, 14:36:49) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import skidl
>>> from skidl import *
>>> search('opamp')
WARNING: Could not open directory ''
WARNING: Non-identical pins with the same number (6) in symbol drawing BAT54ADW
WARNING: Non-identical pins with the same number (3) in symbol drawing BAT54ADW
WARNING: Non-identical pins with the same number (3) in symbol drawing BAV99
WARNING: Non-identical pins with the same number (6) in symbol drawing Rohm_UMP11N
WARNING: Non-identical pins with the same number (3) in symbol drawing Rohm_UMP11N
WARNING: Non-identical pins with the same number (2) in symbol drawing Rohm_UMN1N
WARNING: Non-identical pins with the same number (2) in symbol drawing Rohm_UMN1N
WARNING: Non-identical pins with the same number (2) in symbol drawing Rohm_UMN1N
                                                                  WARNING: Non-identical pins with the same number (4) in symbol drawing MC3486N
WARNING: Non-identical pins with the same number (8) in symbol drawing MC3486N
WARNING: Non-identical pins with the same number (16) in symbol drawing MC3486N
WARNING: Non-identical pins with the same number (8) in symbol drawing MC3486N
WARNING: Non-identical pins with the same number (16) in symbol drawing MC3486N
WARNING: Non-identical pins with the same number (8) in symbol drawing MC3486N
WARNING: Non-identical pins with the same number (12) in symbol drawing MC3486N
WARNING: Non-identical pins with the same number (16) in symbol drawing MC3486N
WARNING: Non-identical pins with the same number (4) in symbol drawing MC3487DX
WARNING: Non-identical pins with the same number (8) in symbol drawing MC3487DX
WARNING: Non-identical pins with the same number (16) in symbol drawing MC3487DX
WARNING: Non-identical pins with the same number (8) in symbol drawing MC3487DX
WARNING: Non-identical pins with the same number (16) in symbol drawing MC3487DX
WARNING: Non-identical pins with the same number (8) in symbol drawing MC3487DX
WARNING: Non-identical pins with the same number (12) in symbol drawing MC3487DX
WARNING: Non-identical pins with the same number (16) in symbol drawing MC3487DX
WARNING: Non-identical pins with the same number (4) in symbol drawing MC3487N
WARNING: Non-identical pins with the same number (8) in symbol drawing MC3487N
WARNING: Non-identical pins with the same number (16) in symbol drawing MC3487N
WARNING: Non-identical pins with the same number (8) in symbol drawing MC3487N
WARNING: Non-identical pins with the same number (16) in symbol drawing MC3487N
WARNING: Non-identical pins with the same number (8) in symbol drawing MC3487N
WARNING: Non-identical pins with the same number (12) in symbol drawing MC3487N
WARNING: Non-identical pins with the same number (16) in symbol drawing MC3487N
WARNING: Non-identical pins with the same number (1) in symbol drawing UA9637
WARNING: Non-identical pins with the same number (4) in symbol drawing UA9637
WARNING: Non-identical pins with the same number (1) in symbol drawing UA9638CD
WARNING: Non-identical pins with the same number (4) in symbol drawing UA9638CD
WARNING: Non-identical pins with the same number (1) in symbol drawing UA9638CP
WARNING: Non-identical pins with the same number (4) in symbol drawing UA9638CP
                     WARNING: Non-identical pins with the same number (AG19) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AH23) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AJ20) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AJ22) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AJ34) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AK13) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AK14) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AK23) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AK35) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AL22) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AM14) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AM22) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AN23) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AP20) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AP22) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AR23) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AM33) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AM34) in symbol drawing P4080-BGA1295
WARNING: Non-identical pins with the same number (AM36) in symbol drawing P4080-BGA1295
[... 96 lines cut ...]
Amplifier_Operational.lib: OPA890xDBV (Single Low-Power, Wideband, Voltage-Feedback Operational Amplifier with Disable, SOT-23-6)
Amplifier_Operational.lib: LM675 (Power Operational Amplifier, TO-220-5)
Amplifier_Operational.lib: NCS2325DM (Dual operational amplifier, 50uV Offset, 0.25uV/C, 35uA Zero-Drift, MSOP-8)
Amplifier_Operational.lib: OPA2325 (Precision, 10-MHz, Low-Noise, Low-Power, RRIO, CMOS Operational Amplifier, SOIC-8)
Amplifier_Operational.lib: MAX4238ASA (Ultra-Low Offset/Drift, Low-Noise, Precision Amplifiers, SOIC-8)
Amplifier_Operational.lib: OPA2695xRGT (Dual, Ultra-Wideband, Current-Feedback Operational Amplifier with Disable, QFN-16)
Amplifier_Operational.lib: ADA4807-2ACP (Dual Rail-to-Rail Input/Output Amplifiers, with disable, SC-70-6/SOT-23-6)
Amplifier_Operational.lib: ADA4870ARRZ (High Speed, High Voltage, 1A Output Drive Amplifier, PSOP-20)
Amplifier_Operational.lib: LM6361 (Single High Speed Operational Amplifier, DIP-8/SOIC-8)
Amplifier_Operational.lib: MCP601-xST (Single 2.7V to 6.0V Single Supply CMOS Op Amps, TSSOP-8)
Amplifier_Operational.lib: LM2902 (Quad Rail-to-rail, 20 MHz GBP, operational amplifiers, SOIC-14/TSSOP-14)
Amplifier_Operational.lib: MCP6L91T-EOT (Single rail-to-rail input/output 8 MHz operational amplifiers, SOT-23-5)
Amplifier_Operational.lib: ADA4807-1 (Single Rail-to-Rail Input/Output Amplifiers, with disable, SC-70-6/SOT-23-6)
Amplifier_Operational.lib: MCP603-xP (Single 2.7V to 6.0V Single Supply CMOS Op Amps, with Chip Select, TSSOP-8)
Amplifier_Operational.lib: OPA333xxD (Single rail-to-rail input/output 8 MHz operational amplifiers, SOIC-8)
Amplifier_Operational.lib: L272D (Dual Power Operation Amplifier, SOIC-16)
Amplifier_Operational.lib: LTC6362xDD (Precision, Low Power, Rail-to-Rail Input/Output, Differential Op Amp/SAR ADC Driver, DFN-8)
Amplifier_Video.lib: AD813 (Single Supply, Low Power, Triple Video Amplifier, DIP-14/SOIC-14)
Device.lib: Opamp_Dual_Generic (Dual operational amplifier)
Device.lib: Opamp_Quad_Generic (Quad operational amplifier)
Power_Management.lib: TSM103W (Dual Operational Amplifier and 2.5V 0.4% Voltage Reference, SOIC-8)
Power_Management.lib: TSM102 (Dual Operational Amplifier and Dual Comparator and 0.4% Voltage Reference, SOIC-16)
Simulation_SPICE.lib: OPAMP (Operational amplifier, single, node sequence=1:+ 2:- 3:OUT 4:V+ 5:V-)
pspice.lib: OPAMP (OPAmp symbol for simulation only)
>>>

xesscorp, 2019-06-14:
The warnings are about library symbols that have more than one pin with the same pin number. I'm not sure why some symbols are designed that way, but it seems like a mistake.

The initial invocation of `search()` is slow (takes over a minute on my PC) because it's scanning in all the libraries (mostly these large FPGA libs) looking for a match with your search string. Once that's done, any other searches are fast (and the warning messages no longer appear). The initial search might be made faster (it's on my to-do list).

xesscorp, 2019-06-15:
I pushed a change that reduced my search() time from 108s to 21s. You can try it to see if it helps you.

xesscorp, 2019-07-11:
OK, guess we're done here.

closed 2019-07-11 by xesscorp