devbisme/skidl #60: Bussed part pins syntax broken for bus length > 9

vegecode, 2019-07-08:
Hello,

Loving skidl, but having an issue with creating a 16 bit wide bus.

I have a microcontroller with a series of pins named FMC_D0 through FMC_D15.

The following returns a list of length 22:
```python
mcu['FMC_D[0:15]']
```

The pins 10-15 are being returned twice, and not in the proper order.

The following returns the same as above:

```python
mcu['FMC_D[0:9]']
```

Modifying it to the following only returns only two exact pins:

```python
mcu['FMC_D[0:9]\W']
```

Am I doing this incorrectly?

I can paste the .lib file if needed. I can confirm now that there is only one of each pin in the file. There isn't more than one pin with `FMC_D11` in the name for example.

vegecode, 2019-07-08:
A suggestion would be to break up the numbers on either side of the colon and then modify the regex to be each number followed by a non-word character.

```python
mcu['FMC_D[0:15]']
```

would search consecutively for:

```python
mcu['FMC_D0\W']
mcu['FMC_D1\W']
...
mcu['FMC_D15\W']
```

vegecode, 2019-07-08:
Hmmm...

As I moved on I realized that the syntax works fine for another part with pin names: `DQ[0:15]`

xesscorp, 2019-07-08:
Can you send me the .lib file with the MCU part that is causing the problem?

vegecode, 2019-07-09:
[problemlib.zip](https://github.com/xesscorp/skidl/files/3374473/problemlib.zip)

Licensing thing: the two items in there are from snapEDA which has a Creative Commons Share-Alike license.

```
MCU = Part(
    'GameteSnapEDA',
    'STM32F767ZGT6',
    dest=TEMPLATE,
)
mcu = MCU()
```

Thanks for looking at this.

xesscorp, 2019-07-09:
You were on the right track: the matching algorithm was too aggressive. I modified it to add a regex for a non-number at the end of the matching string. That seems to have fixed the problem exposed by your library. Try it and see if it works for you:

    pip install -U git+https://github.com/xesscorp/skidl

vegecode, 2019-07-10:
That fixes the one case. Another case in the same file is now failing:

```python
sdram = Part(
'GameteSnapEDA',
'MT48LC16M16A2TG-6A_IT:GTR')

sdram['DQ[0:15]']
```

The search returns nothing. I think maybe the regex could be something like '[\D$]' so that the end of the string/line will match, too.

xesscorp, 2019-07-10:
OK, sorry. We're playing regex whack-a-mole. I've committed a new fix and tested it with both your MCU and RAM with both increasing and decreasing bit orders. Seems to work. Give it a try and let me know.

vegecode, 2019-07-10:
Now this might just be me doing it wrong:

```python
In [18]: sdram['A[0:12]']
Out[18]: 
[Pin U3/23/A0/INPUT,
 Pin U3/20/BA0/INPUT,
 Pin U3/24/A1/INPUT,
 Pin U3/21/BA1/INPUT,
 Pin U3/25/A2/INPUT,
 Pin U3/26/A3/INPUT,
 Pin U3/29/A4/INPUT,
 Pin U3/30/A5/INPUT,
 Pin U3/31/A6/INPUT,
 Pin U3/32/A7/INPUT,
 Pin U3/33/A8/INPUT,
 Pin U3/34/A9/INPUT,
 Pin U3/22/A10/INPUT,
 Pin U3/35/A11/INPUT,
 Pin U3/36/A12/INPUT]

In [19]: sdram['^A[0:12]']
Out[19]: 
[Pin U3/23/A0/INPUT,
 Pin U3/24/A1/INPUT,
 Pin U3/25/A2/INPUT,
 Pin U3/22/A10/INPUT,
 Pin U3/35/A11/INPUT,
 Pin U3/36/A12/INPUT]

In [20]: sdram['[^B]A[0:12]'] # no out field

In [21]: sdram['^A[0:12]$']
Out[21]: [Pin U3/23/A0/INPUT, Pin U3/24/A1/INPUT, Pin U3/25/A2/INPUT]
```

How do I exclude the 'BA0' and 'BA1'?

xesscorp, 2019-07-10:
OK, more regex whack-a-mole. The problem with sdram['^A[0:12]'] is that SKiDL triggers the bus slices when it sees a bus name followed by the bracketed indices. The leading '^' prevents that trigger and SKiDL goes back to looking at it as a pure regular expression. In that case, it would look for ^A0, ^A:, ^A1 and ^A2 which is why it found A0, A1, A2, A10, A11 and A12.

I adjusted SKiDL so it triggers the bus slicing whenever bracketed indices are detected. I'm not sure it won't cause a problem later, but it works for all your examples and it passes the existing unit tests.

vegecode, 2019-07-11:
Thanks so much! It appears to be working for my project now.

closed 2019-07-11 by vegecode

xesscorp, 2019-07-11:
I had to make a correction to the regex. Go ahead and re-install. It should work the same for you.