devbisme/skidl #61: Apply external power to net to satisfy ERC

pepijndevos, 2019-07-12:
I have a circuit with some input pins and power supplied externally.
So ERC thinks there is nothing that drives these nets and complains.
I don't want to completely disable ERC for that net, I just want to tell it I will connect something to it.

In Kicad there is `PWR_FLAG`, which silences the warning, but then of course when you make a netlist it complains it has no footprint.

Is there a way to do this?

xesscorp, 2019-07-12:
There are a number of methods:
1. Attach a no-connect to the input pins: `in_pin += NC`.
2. Create a net, set its drive level to POWER, and attach it to the unconnected power inputs:
```py
        a = Net()
        a.drive = POWER
        my_part[pwr_pin] += a
```
3. Turn off ERC checking for a particular pin: `my_part[my_pin].do_erc = False`

pepijndevos, 2019-07-12:
Thank you!

It's a bit magic where the NC comes from when I don't do a star import, so I expected `skidl.NC`, but it works. Might be useful to document this somewhere.

closed 2019-07-12 by pepijndevos