devbisme/skidl #62: KiCad Connection Problems With New Installation

gfleetwood, 2019-07-14:
Hi,

I just installed skidl and I can't get the example in the README to work.

```
from skidl import *

# Create input & output voltages and ground reference.
vin, vout, gnd = Net('VI'), Net('VO'), Net('GND')

# Create two resistors.
r1, r2 = 2 * Part('device', 'R', TEMPLATE, footprint='Resistor_SMD.pretty:R_0805_2012Metric')

....
....

generate_netlist()
```

I'm on Ubuntu 18.04. Here's the stack trace.

```
ERROR: Can't open file: device.

WARNING: Could not load KiCad schematic library "device", falling back to backup library.
Traceback (most recent call last):
  File "/home/gordon/anaconda3/lib/python3.6/site-packages/skidl/tools/kicad.py", line 69, in _load_sch_lib_
    f, _ = find_and_open_file(filename, lib_search_paths_, lib_suffixes[KICAD])
  File "/home/gordon/anaconda3/lib/python3.6/site-packages/skidl/utilities.py", line 263, in find_and_open_file
    raise FileNotFoundError
FileNotFoundError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 7, in <module>
    r1, r2 = 2 * Part('device', 'R', TEMPLATE, footprint='Resistor_SMD.pretty:R_0805_2012Metric')
  File "/home/gordon/anaconda3/lib/python3.6/site-packages/skidl/Part.py", line 139, in __init__
    raise e
  File "/home/gordon/anaconda3/lib/python3.6/site-packages/skidl/Part.py", line 131, in __init__
    lib = SchLib(filename=libname, tool=tool)
  File "/home/gordon/anaconda3/lib/python3.6/site-packages/skidl/SchLib.py", line 96, in __init__
    load_func(filename, skidl.lib_search_paths[tool])
  File "/home/gordon/anaconda3/lib/python3.6/site-packages/skidl/tools/kicad.py", line 73, in _load_sch_lib_
    filename, str(e)))
Exception: Unable to open KiCad Schematic Library File device ()
```

xesscorp, 2019-07-14:
What version of KiCad are you using? Older versions use devices.lib, while newer versions use device.lib. Have you looked in your KICAD_SYMBOL_DIR to see if the device.lib file is present?

gfleetwood, 2019-07-14:
I'm using version 5.1.2. It was "Device.lib" in KICAD_SYMBOL_DIR  so once I changed that the code worked. Thanks!

closed 2019-07-14 by gfleetwood

xesscorp, 2019-07-14:
OK, I guess it's case-sensitive on linux. Thanks for letting me know!

matthuszagh, 2019-12-20:
I'm getting the same issue here (Device.lib instead of device.lib). Would it be possible to either make the test case-insensitive, or change the name to look for the capitalized version? The current kicad master branch uses Device.lib.

matthuszagh, 2019-12-20:
I have a fix for this, but it requires iterating through each file in each path and comparing the lowercase filenames. This might add too much overhead, I'm not sure. But I'm also not sure how else to do it, other than to require case-sensitive library names. Any thoughts? Happy to show my code if you'd like.

matthuszagh, 2019-12-20:
On a only slightly related note, I have to add the following line before `xess_lib=` in order to get the rest of the tests to work:
```python
lib_search_paths[KICAD].append(os.path.dirname(os.path.abspath(__file__)))
```
This is because `make test` is run from the top level directory, which is added to the library lookup paths, but the "tests" dir isn't.

xesscorp, 2019-12-20:
I changed "device" to "Device" because that's the official name of the library in the KiCad distribution. That should fix the problem when using a case-sensitive OS. Let me know if there are further problems.

matthuszagh, 2019-12-20:
Terrific, works for me. Thanks for such a quick fix! Did you see my last comment about lib_search_paths for tests?

xesscorp, 2019-12-21:
I don't use the `make test` command. It calls `python setup.py test` and that expects a `unittest` framework. Also, I get a warning that using the `setup.py test` is deprecated.

For testing, you can use `make test-all` or run `tox` directly. You can also use `pytest tests` if you just want to run the tests for the current Python version.

I suppose I could change the makefile to make it run `pytest tests`...

matthuszagh, 2019-12-21:
That seems like it could be a reasonable solution.

I'm getting a few errors when running `pytest tests`:

`test_backup_1`: `FileNotFoundError: Can't open file: crap.`

`test_part_from_non_existing_lib_cannot_be_instantiated`: `FileNotFoundError: Can't open file: non-existing.`

`test_lib_import_1`: `FileNotFoundError: Can't open file: lt1083.`
It seems here that you've placed lt1083 in a personal,
windows-specific directory. Is it possible to bundle the lt1083 spice
file in the tests/ directory, or does that violate some licensing
issue?

`test_lib_export_1`: same issue, related to lt1083