devbisme/skidl #63: Autocompletion (Definitely wishlist!)

robot-army, 2019-07-27:
I'm so in love with skidl so far! One thing I'd really love to see is some kind of autocompletion (maybe for visual studio code?) that knows about libraries and footprints.

xesscorp, 2019-07-28:
I'm glad your enjoying SKiDL! Making it easier to find and instantiate parts is a good idea. Autocompletion is one way to do that, but wouldn't that have to be specifically coded for each particular editor? (For example, I could put autocompletion into Notepad++, but a different implementation would be needed for Visual Studio, right?)

As a first attempt, I'm going to make a small wxpython app that will let you enter a search string for a part and then it will present a list of all the libraries and parts that match the query. You'll be able to copy the Part(...) instantiation onto the clipboard and paste it into your file. You can try this app and maybe it will spark some ideas for how we can support a more integrated version.

robot-army, 2019-07-28:
That sounds super helpful. I'm not sure how the various editors do their
autocompletion, I'm sure there's a common approach or framework, but I
wasn't able to find much in a very quick glance.

That'd be super cool, probably more helpful, especially if it gave a
preview of the pins names and numbers. Also, bring back cvpcb!! :)
Seriously though, something similar for footprints would be sweet.

Nice work!

robot-army, 2019-07-30:
I was wondering, would it be easier to pinch KiCAD's existing tools for this, and just add a 'copy path to thingie' button?

xesscorp, 2019-07-31:
That's a good idea! It eliminates a bunch of work. I was hoping the symbol editor was a free-standing executable, but it appears you can only run it when KiCad is running.

That said, I do have a version of a wxpython app that does some of what we want. I'll upload it to the SKiDL repo later today and tell you how to install it. You can play around with it and see what features you like, dislike, or are missing. That may guide us as to how to proceed with either this tool or by modifying the KiCad symbol editor.

[image]

xesscorp, 2019-07-31:
I uploaded the part search GUI to Github. You can install it like this:

`pip install -U git+https://github.com/xesscorp/skidl`

You can start the part search utility using the command `skidl_part_search`. It will take a few seconds to do the first search as it scans all the library files. Any searches after that should be quick. (If we move forward with this utility, it will need some means for specifying the list of directories it looks in for symbol files.)

robot-army, 2019-08-02:
Hihi! I'm ... struggling to install this on a fresh copy of Ubuntu. I've tried a few things but I'm not having much luck :(

Any tips on getting the environment the way it should be? I'm sure it's something simple I'm missing...

[skidl_install_issues.txt](https://github.com/xesscorp/skidl/files/3459449/skidl_install_issues.txt)

xesscorp, 2019-08-02:
That's a lot of text. I'll assume you're having some problem getting wxpython to install or something.
Here's what I did on a new install of Ubuntu 18.04:
```bash
>>> sudo apt install git make gcc libgtk-3-dev libwebkitgtk-dev libwebkitgtk-3.0-dev libgstreamer-gl1.0-0 freeglut3 freeglut3-dev python-gst-1.0 python3-gst-1.0 libglib2.0-dev ubuntu-restricted-extras libgstreamer-plugins-base1.0-dev

>>> pip install git+https://github.com/xesscorp/skidl
```
After that, I was able to get `skidl_part_search` running, but it didn't really do anything because there were no KiCad libraries to scan.

vegecode, 2019-08-04:
After this update, I can no longer install Skidl.

Every install fails:

```
...
    warning: no previously-included files matching '__pycache__' found under directory 'wx'
    writing manifest file 'wxPython.egg-info/SOURCES.txt'
    copying wx/_adv.cpython-36m-x86_64-linux-gnu.so -> build/lib.linux-x86_64-3.6/wx
    error: could not write to 'build/lib.linux-x86_64-3.6/wx/_adv.cpython-36m-x86_64-linux-gnu.so': No space left on device
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-_jmqeaou/wxpython/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-5s9uj7qp-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-_jmqeaou/wxpython/
```

Could the wxpython app be in a different repository? I find that using ipython works fine for searching for parts. You can also grep around in the KiCad folders directly or in the skidl libraries that ship with Skidl.

xesscorp, 2019-08-04:
Are you trying to install this into the Python that comes bundled with KiCad? I'm not sure what problems that might cause.

I would try installing into the Python 2 or 3 that comes with your linux dist.

I've previously used a Python window running search to search for parts. Something a little more integrated might be nice.

vegecode, 2019-08-04:
I am using the version of pip that came with:

```
sudo apt install python3-pip
```

I'll just have to figure out the error I guess.

Is searching the parts that come with KiCad the way that you discover parts to use in your designs? What do you use the search for?

I used it at first until I understood that Skidl was finding the libraries as I expected. Then I started looking into the library files themselves since they are plain text and contain the same information.

vegecode, 2019-08-04:
```
TMPDIR=/home/alexander/.tmp/ pip3 install --cache-dir=/home/alexander/.tmp/ --build /home/alexander/.tmp/ -U git+https://github.com/xesscorp/skidl
```

I got it to build by making a different tempdir in my home folder and passing that in for the options. My /tmp partition is 3.6 GB which isn't enough surprisingly.

xesscorp, 2019-08-04:
Good catch! I never would have found that. 

Looking directly into the .lib files is one way to find parts, but I like the keyword search because it can help find parts that are in places you wouldn't expect. For example, searching for "ADC" finds a bunch of ADC chips, but also pulls out processors with integrated ADCs from several different MCU family libraries.

johnthagen, 2019-10-19:
Perhaps slightly off topic, but if you use [type hinting](https://docs.python.org/3/library/typing.html), editors like [PyCharm](https://www.jetbrains.com/help/pycharm/type-hinting-in-product.html) can autocomplete for the user as they type.

Note, this applies to Python types not data, so I'm not sure this would apply to footprints and libraries (unless they were represented as Python types).

closed 2021-07-09 by xesscorp