devbisme/skidl #64: Combining Kicad and Skidl part libraries

vegecode, 2019-08-07:
I am not sure what I am doing wrong.

My default tool is KICAD and my search paths are like so:

```python
lib_search_paths[KICAD].append('../library/snapEDA')
lib_search_paths[SKIDL].append('../library/skidl_lib')
lib_search_paths[KICAD].append('/usr/share/kicad/library')
```

in the SKIDL search path the part definition is in a file called gamete_lib.py.

The error I get is:

```ERROR: Can't open file: gamete.

WARNING: Could not load KiCad schematic library "gamete", falling back to backup library.
Traceback (most recent call last):
  File "/home/alexander/.local/lib/python3.6/site-packages/skidl/tools/kicad.py", line 60, in _load_sch_lib_
    f, _ = find_and_open_file(filename, lib_search_paths_, lib_suffixes[KICAD])
  File "/home/alexander/.local/lib/python3.6/site-packages/skidl/utilities.py", line 290, in find_and_open_file
    logger, FileNotFoundError, "Can't open file: {}.\n".format(filename)
  File "/home/alexander/.local/lib/python3.6/site-packages/skidl/utilities.py", line 820, in log_and_raise
    raise exc_class(message)
FileNotFoundError: Can't open file: gamete.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "gamete.py", line 123, in <module>
    footprint='Package_TO_SOT_SMD:SOT-353_SC-70-5_HandSoldering',
  File "/home/alexander/.local/lib/python3.6/site-packages/skidl/Part.py", line 141, in __init__
    raise e
  File "/home/alexander/.local/lib/python3.6/site-packages/skidl/Part.py", line 131, in __init__
    lib = SchLib(filename=libname, tool=tool)
  File "/home/alexander/.local/lib/python3.6/site-packages/skidl/SchLib.py", line 99, in __init__
    load_func(filename, skidl.lib_search_paths[tool])
  File "/home/alexander/.local/lib/python3.6/site-packages/skidl/tools/kicad.py", line 64, in _load_sch_lib_
    filename, str(e)
FileNotFoundError: Unable to open KiCad Schematic Library File gamete (Can't open file: gamete.
)
```

The error looks like it is searching for a KICAD libary file. Do I have to change the default tool every time I search for a part that is in the other supported type? Skidl found previous Kicad parts that I instantiated.

vegecode, 2019-08-07:
Trying my own suggestion didn't work either.

I made the repo public and pushed it in it's failing state. I don't understand why the library isn't being found.

To build you have to cd into this directory:
`https://github.com/vegecode/Gamete/tree/master/pcb/skidl`
and use command `python3 gamete.py`

Can someone help me understand why i am getting the above error about the file not being found?

xesscorp, 2019-08-08:
I tried your code and found these errors:
1. Instead of calling your library `gamete_lib.py`, you'll need to call it `gamete_sklib.py`.
2. When you access the library, you'll have to do it like this:
```python
led_driver = Part(
        'gamete',
        'TPS61169',
        tool=SKIDL,
        manu='Texas Instruments',
        part_no='TPS61169DCKR',
        footprint='Package_TO_SOT_SMD:SOT-353_SC-70-5_HandSoldering',
)
```

   or like this:

```python
led_driver = Part(
        'gamete_sklib.py',
        'TPS61169',
        tool=SKIDL,
        manu='Texas Instruments',
        part_no='TPS61169DCKR',
        footprint='Package_TO_SOT_SMD:SOT-353_SC-70-5_HandSoldering',
)
```
3. In `gamete_sklib.py`, you'll need to change pin functions from things like `Pin.PWROUT` to `Pin.types.PWROUT`. (I'll need to fix SKiDL to allow the shorter version of pin functions.)
4. On line 125 of `gamete.py`, prepend `gen.` to `Diode_SMD` to access the part inside the `generic_parts` file.

After making these changes, I was able to get your code to compile. There were a lot of unconnected pin warnings and such, but you probably expected those.

vegecode, 2019-08-08:
>    1. In gamete_sklib.py, you'll need to change pin functions from things
>    2. On line 125 of gamete.py, prepend gen. to Diode_SMD to access the

Thank you very much.

I figured there was a naming convention I wasn't aware of.

vegecode, 2019-08-08:
https://github.com/xesscorp/skidl/blob/729eb362997228dbeb0c81a529920b064662cfcd/skidl/libs/74xgxx_sklib.py#L9

About the `Pin.types.PWROUT`, are these files just bit-rotted?

closed 2019-08-23 by vegecode