devbisme/skidl #65: Read field values of KiCad parts into Skidl definition

vegecode, 2019-08-23:
Hello,

I have a personal library of parts in a .lib file with the fields all filled out including the footprint, manufacturer name and no, distributor part no, etc.

It would be really great if Skidl could read in that information into fields when instantiating a part. I have to include that information into my KiCad library in order for KiCad to have the correct footprint for all my part symbols so the work is already done once.

Here's an example from my parts-lib.lib:
```
#
# NSR0240
#
DEF NSR0240 D 0 40 N N 1 F N
F0 "D" 0 100 50 H V C CNN
F1 "NSR0240" 0 -100 50 H V C CNN
F2 "Diode_SMD:D_SOD-323" 50 -250 50 H I L CIN
F3 "https://datasheet.lcsc.com/szlcsc/ON-Semicon-ON-NSR0240HT1G_C152519.pdf" -150 -50 50 H I C CNN
F4 "NSR0240HT1G" 210 270 50 H I C CNN "Part No."
F5 "ON Semiconductor" 110 170 50 H I C CNN "Manufacturer"
F6 "LCSC" 110 170 50 H I C CNN "Distributor"
F7 "C152519" 110 170 50 H I C CNN "Distributor Part No."
F8 "Schottky (250mA Continuous Current) (.48V FVD) (.2uA @ 25V Reverse)" 110 170 50 H I C CNN "Description"
DRAW
P 2 0 1 0 50 0 -50 0 N
P 4 0 1 8 50 50 50 -50 -50 0 50 50 N
P 6 0 1 8 -75 25 -75 50 -50 50 -50 -50 -25 -50 -25 -25 N
X K 1 -150 0 100 R 50 50 1 1 P
X A 2 150 0 100 L 50 50 1 1 P
ENDDRAW
ENDDEF
```

As you can see, the footprint is filled out, and with the LCSC order number, this library file is effectively an atomic library, I think it's called?

I was even thinking of adding an "in-stock" field to keep count of how many I have left. It would be easy to integrate a BOM generation tool that way.

F2 is always the footprint, F3 is always the datasheet, and then the other fields are the optional extra ones you can add but must name.

xesscorp, 2019-08-25:
The information you want is already stored in the Part structure. For example:
```python
In [1]: from skidl import *

In [2]: nsr = Part('test','NSR0240')

In [3]: nsr.fields
Out[3]:
[{'reference': '"D"',
  'posx': '0',
  'posy': '100',
  'text_size': '50',
  'text_orient': 'H',
  'visibility': 'V',
  'htext_justify': 'C',
  'vtext_justify': 'CNN'},
 {'name': '"NSR0240"',
  'posx': '0',
  'posy': '-100',
  'text_size': '50',
  'text_orient': 'H',
  'visibility': 'V',
  'htext_justify': 'C',
  'vtext_justify': 'CNN',
  'fieldname': ''},
 {'name': '"Diode_SMD:D_SOD-323"',
  'posx': '50',
  'posy': '-250',
  'text_size': '50',
  'text_orient': 'H',
  'visibility': 'I',
  'htext_justify': 'L',
  'vtext_justify': 'CIN',
  'fieldname': ''},
 {'name': '"https://datasheet.lcsc.com/szlcsc/ON-Semicon-ON-NSR0240HT1G_C152519.pdf"',
  'posx': '-150',
  'posy': '-50',
  'text_size': '50',
  'text_orient': 'H',
  'visibility': 'I',
  'htext_justify': 'C',
  'vtext_justify': 'CNN',
  'fieldname': ''},
 {'name': '"NSR0240HT1G"',
  'posx': '210',
  'posy': '270',
  'text_size': '50',
  'text_orient': 'H',
  'visibility': 'I',
  'htext_justify': 'C',
  'vtext_justify': 'CNN',
  'fieldname': '"Part No."'},
 {'name': '"ON Semiconductor"',
  'posx': '110',
  'posy': '170',
  'text_size': '50',
  'text_orient': 'H',
  'visibility': 'I',
  'htext_justify': 'C',
  'vtext_justify': 'CNN',
  'fieldname': '"Manufacturer"'},
 {'name': '"LCSC"',
  'posx': '110',
  'posy': '170',
  'text_size': '50',
  'text_orient': 'H',
  'visibility': 'I',
  'htext_justify': 'C',
  'vtext_justify': 'CNN',
  'fieldname': '"Distributor"'},
 {'name': '"C152519"',
  'posx': '110',
  'posy': '170',
  'text_size': '50',
  'text_orient': 'H',
  'visibility': 'I',
  'htext_justify': 'C',
  'vtext_justify': 'CNN',
  'fieldname': '"Distributor Part No."'},
 {'name': '"Schottky (250mA Continuous Current) (.48V FVD) (.2uA @ 25V Reverse)"',
  'posx': '110',
  'posy': '170',
  'text_size': '50',
  'text_orient': 'H',
  'visibility': 'I',
  'htext_justify': 'C',
  'vtext_justify': 'CNN',
  'fieldname': '"Description"'}]
```
Admittedly, the information is a bit hard to get out: you'll have to search the `fields` list for the element having the `fieldname` that you want. Maybe I should create a dictionary that uses the `fieldname` as the key and stores the `name` contents as the value. Or maybe you have some other idea.

xesscorp, 2019-12-18:
Every `Part` object now stores field information in a dictionary called `fields`:

```terminal
>> lm35 = Part('Sensor_Temperature', 'LM35-D')
>> lm35.fields

{'F0': 'U',
 'F1': 'LM35-D',
 'F2': 'Package_SO:SOIC-8_3.9x4.9mm_P1.27mm',
 'F3': ''}

>>> lm35.fields['F1']

'LM35-D'
```

Fields can also be accessed as attributes of the part:

```terminal
>>> lm35.F1

'LM35-D'
```

New fields can be added just by adding new keys and values to the `fields` dictionary.

closed 2019-12-18 by xesscorp