devbisme/skidl #66: zyc suggestions

AppMathIsRealMath, 2019-10-07:
You've created another great tool - zyc is very, very convenient.

A couple of suggestions:
-- I use a dark theme and the search results are invisible unless I highlight them. To get around that I use "env GTK_THEME=Adwaita:light zyc" but either handling light and dark themes or allowing us to choose font colors would be nice.
-- I use Liberation Mono with a fairly large font size and the various buttons (esp the search button) clip their text - automatically resizing the buttons to account for different text sizes would also be nice. The "Searching..." popup text also clips.

I can provide a screenshot if that would help.

xesscorp, 2019-10-08:
Thanks for the feedback!

I've updated the repo with these changes:

1. The luminance of the system background color is now used to detect dark mode and the grid table cell color is set accordingly.
2. The fixed size of the Search and Copy buttons was removed so they should adjust to the size of the font.

I didn't make any changes to the Searching progress bar since it was called with the default parameters. If it can't handle different font sizes without clipping, then that's a problem they'll have to handle.

Give the updated repo a try and let me know if it fixes any of your problems.

AppMathIsRealMath, 2019-10-09:
It looks like your changes recognize the dark theme and font size now. The Search button text is now unclipped - I realized later I could resize the Searching popup window to display more text if desired.

However the search result text is still invisible unless highlighted. I'm using Flat-Plat-Dark as my gtk theme on Linux if that helps you test changes (don't know what system you develop on).

I'll attach some screenshots:

![zyc0](https://user-images.githubusercontent.com/47261720/66457271-a2fa7500
ea35-11e9-87a8-688397e99364.png)
[image]

AppMathIsRealMath, 2019-10-09:
It looks like your changes recognize the dark theme and font size now. The Search button text is now unclipped - I realized later I could resize the Searching popup window to display more text if desired.

However the search result text is still invisible unless highlighted. I'm using Flat-Plat-Dark as my gtk theme on Linux if that helps you test changes (don't know what system you develop on).

I'll attach some screenshots:

[image]
[image]

AppMathIsRealMath, 2019-10-09:
BTW, since this is just a cosmetic issue and not functional - and easily worked around by setting the GTK_THEME env variable in an alias - don't knock yourself out trying to fix weird theme interactions.

I've had similar problems with the st-link gui and even the "meld" diff tool that I worked around. I'd rather you work on cool new tools and functionality!

Thanks for looking into it.

xesscorp, 2019-10-09:
Your screen shots showed that zyc was not correctly detecting dark mode (since the table cells were still white and not black). I corrected my luminance calculation and I think it should work now. Let me know.

AppMathIsRealMath, 2019-10-09:
Works correctly now, thanks! 

There is one oddity where the part and a newly selected associated footprint turns the part cells either white or black (depending on theme, see screenshots) and vice versa. Perhaps showing the association between part and a selected footprint could be a little less obtrusive.

Screenshots:

[image]

[image]

xesscorp, 2019-10-10:
I'm getting the same results as you in dark mode. In light mode, I'm *not* getting that blacked-out line in the Part Search Pane and everything is readable. I'm using the most recent release of Pop! OS and wxpython.version() gives "4.0.6 gtk3 (phoenix) WxWidgets 3.0.5". My libgtk is 3.24.8-1.

I've tried a lot of program changes, including just using the default wxpython cell colors with no modifications for light/dark mode. In linux dark mode, they all have that same white-out of the selected line in the tables. Light mode in linux works, and light/dark modes both work in Windows.

AppMathIsRealMath, 2019-10-10:
I'm using Solus Linux (a "curated" rolling release distro with the Budgie desktop). My libgtk is 3.24.11. I'm running a python3 virtualenv with wx.version() is "4.0.6 gtk3 (phoenix) wxWidgets 3.0.5".

When I tried theme Adwaita:light the selected line was normal - no black-out, as you reported for light themes. The light theme screenshot above that showed the black-out (although text is dimly visible) was Flat-Plat.

So I'm guessing we're deep in the briar patch of  Linux desktop environments/distros/theming variations... ;-)

AppMathIsRealMath, 2019-10-10:
Dammit, I did it again - hit the close&comment button instead of the big glowing green comment button...

*sigh*

Feel free to close the issue - the original text clipping problem is working just fine now and any theming issues can be worked around without any trouble.

Thanks!

xesscorp, 2019-10-11:
OK, I'll close this since you at least have something that works.

closed 2019-10-11 by xesscorp