devbisme/skidl #71: Invalid escape sequence in part_query.py

johnthagen, 2019-12-09:
```
/home/travis/build/xesscorp/skidl/skidl/part_query.py:329
  /home/travis/build/xesscorp/skidl/skidl/part_query.py:329: DeprecationWarning: invalid escape sequence \(
    set(re.findall("\(\s*pad\s+([^\s)]+)", " ".join(module_text)))
```

xesscorp, 2019-12-09:
I'm confused by this. I want to match the literal "(" character at the start of the string, not start a group. All the man pages for the `re` module (from 2.7 up to 3.8) state that I can use "\(" or "[(]" to do that. So why is this flagged as deprecated? Is this a spurious error message, or am I overlooking something?

johnthagen, 2019-12-09:
@xesscorp I think the correct what do create regular expression strings to avoid any of these issues is to use a raw string: `r"(\s*pad\s+([^\s)]+)"`, that avoids the issue of having to distinguish between escaping in the string and using `\` in the regular expression string.

See https://stackoverflow.com/a/50504635/1398841

xesscorp, 2019-12-10:
OK, I see now. I made the change to a raw string and checked it in. Thanks!

closed 2019-12-10 by xesscorp

johnthagen, 2019-12-10:
@xesscorp I think this was fixed in error: 27cb16815537b53a50d15953f35b32c8074cf327

I believe you want to remove the `\` before the first `(` since a raw string will keep the `\` now.

```py
r"\(\s*pad\s+([^\s)]+)
r"(\s*pad\s+([^\s)]+)
```

xesscorp, 2019-12-10:
The regex is trying to detect a starting paren of a pad definition. If I removed the `\`, then the paren would be looked at as the start of a group. That's not what I want. There would also be unbalanced grouping parens: two `(` and only one `)`.

johnthagen, 2019-12-10:
@xesscorp Ah, I see what you're saying. You're right, you did it correctly. 👍

xesscorp, 2019-12-10:
Even a blind squirrel finds a nut once and a while!