devbisme/skidl #80: could not install downloaded tar-gz from pypi without network connection

Aydan, 2020-03-31:
I could not install with `pip3 install skidl-0.0.29.tar.gz` although all dependencies were already installed because installed packages are ignored by the installer:

```
  ERROR: Command errored out with exit status 1:
   command: '[snip]\programs\python\python37-32\python.exe' '[snip]\programs\python\python37-32\lib\site-packages\pip' install --ignore-installed --no-user --prefix '[snip]\AppData\Local\Temp\24\pip-build-env-ov8u9upn\overlay' --no-warn-script-location --no-binary:none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=40.8.0' wheel 
```
because "--ignore-installed" is used and pip3 tries to redownload setuptools-46.1.3-py3-none-any.whl although it is already installed.

xesscorp, 2020-05-12:
Is there a reason you couldn't just use `pip install skidl`? Or if you wanted to install from the repo: `pip install git+https://github.com/xesscorp/skidl`.

Aydan, 2020-05-12:
Yes, pip doesn't like our company proxy. And I think it's also bad practice to ignore locally installed dependencies.