devbisme/skidl #81: No Fields for Parts from SKiDL Libraries

jbayless, 2020-04-23:
According to the documentation, Parts should have fields:

> 
> Parts typically have field that store additional information such as manufacturer identifiers. Every Part object stores this information in a dictionary called fields:
> 
> >> lm35 = Part('Sensor_Temperature', 'LM35-D')
> >> lm35.fields
> 
> {'F0': 'U',
>  'F1': 'LM35-D',
>  'F2': 'Package_SO:SOIC-8_3.9x4.9mm_P1.27mm',
>  'F3': ''}
> 
> >>> lm35.fields['F1']
> 
> 'LM35-D'
> 

Parts that are imported from KiCAD libraries do have fields that behave this way. But parts that are created from SKiDL's libraries do not.

```
import skidl as sk
sk.set_default_tool(sk.KICAD)
r1 = sk.Part("Device", 'R', value=500)
print(r1.fields)
```

> {'F0': 'R', 'F1': 'R', 'F2': '', 'F3': ''}

```
import skidl as sk
sk.set_default_tool(sk.SKIDL)
r1 = sk.Part("Device", 'R', value=500)
print(r1.fields)
```

> AttributeError: 'Part' object has no attribute 'fields'

It seems like it would be a good idea to at least have the Part initialize with a 'fields' attribute, even if it's an empty dictionary, so that it at least has a consistent interface regardless of where the Part is loaded from.

xesscorp, 2020-04-23:
Currently, the `fields` attribute only exists in the KiCad parts. It should be moved into the generic `Part` object. I'll address this when I work on issue #82.

xesscorp, 2020-04-25:
I've pushed a fix for #81 and #82 to the development branch. You can install it with

```bash
$ pip install https://github.com/xesscorp/skidl/archive/development.zip
```

I tested it with your examples and some others. I also added the ability to mirror the attributes back into the `fields` dictionary so you can make changed like `r1.F0 = something` and it will be seen in `r1.fields['F0']`.

Let me know if this works for you.

jbayless, 2020-04-26:
Works for me! Thank you for the fast fix!
I think a further improvement would be for the constructor to automatically cast fields as an AttrDict, since it's assumed to behave as one.

Right now:

```
works = sk.SchLib(tool=sk.SKIDL).add_parts(*[
    sk.Part(name = "r",
                   dest = sk.TEMPLATE,
                   tool=sk.SKIDL,
                   keywords='resistor',
                   description='My favourite custom resistor',
                   ref_prefix='R',
                   num_units=1,
                   do_erc=True,
                   pins=[
                       sk.Pin(num='1',name='R1',func=sk.Pin.PASSIVE,do_erc=True),
                       sk.Pin(num='2',name='R2',func=sk.Pin.PASSIVE,do_erc=True)],
                   fields = sk.AttrDict({"my_field": "my value"}))   
    ])
working_r = sk.Part(works, 'r')
print(working_r.my_field)
```

> my value

```
fails = sk.SchLib(tool=sk.SKIDL).add_parts(*[
    sk.Part(name = "r",
                   dest = sk.TEMPLATE,
                   tool=sk.SKIDL,
                   keywords='resistor',
                   description='My favourite custom resistor',
                   ref_prefix='R',
                   num_units=1,
                   do_erc=True,
                   pins=[
                       sk.Pin(num='1',name='R1',func=sk.Pin.PASSIVE,do_erc=True),
                       sk.Pin(num='2',name='R2',func=sk.Pin.PASSIVE,do_erc=True)],
                   fields = {"my_field": "my value"})   
    ])
failing_r = sk.Part(fails, 'r')
print(failing_r.my_field)
```
> AttributeError: 'dict' object has no attribute 'sync'

xesscorp, 2020-04-26:
OK, I changed it so that a dict gets converted to an AttrDict.

jbayless, 2020-04-26:
Great! I'll close the issue.

closed 2020-04-26 by jbayless