devbisme/skidl #82: Part fields cannot be accessed as attributes unless loaded from library

jbayless, 2020-04-23:
According to the documentation:

> Fields can also be accessed as attributes of the part:
> 
> >>> lm35.F1
> 
> 'LM35-D'
> 
> New fields can be added just by adding new keys and values to the fields dictionary.

These statements aren't both simultaneously true, though. Fields cannot be accessed as attributes if they are added just by adding new keys and values to the fields dictionary.

```
import skidl as sk
sk.set_default_tool(sk.KICAD)
r1 = sk.Part("Device", 'R', value=500)
print(r1.fields["F0"])
print(r1.F0)
r1.fields["my_custom_field"]="my custom value"
print(r1.fields["my_custom_field"])
print(r1.my_custom_field)

> 'R'
> 'R'
> 'my custom value'
> AttributeError: 'Part' object has no attribute 'my_custom_field'
```

xesscorp, 2020-04-23:
You're right. The `fields` attribute of a `Part` object needs added functionality. I'll go to work on that.

jbayless, 2020-04-26:
Thanks for fixing this issue!

closed 2020-04-26 by jbayless

xesscorp, 2020-04-27:
No problem! Thanks for reporting the problem and testing the solution. Please report any further problems you encounter.