devbisme/skidl #83: [SKiDL Feature Request] support for Horizon-EDA

AppMathIsRealMath, 2020-05-04:
Have you had a chance to look at [Horizon-EDA](https://github.com/horizon-eda/horizon) here on github? I've been playing around a bit with it and so far like it - especially its support for viewing exploded 3d boards... ;-)

Since using schematic capture makes me break out in hives it would be awesome if SKiDL could be used with Horizon-EDA. Parts are managed in a so-called "pool" and are in JSON format so that hopefully wouldn't be too difficult to integrate. I don't know if the netlist format would pose any difficulties, though.

xesscorp, 2020-05-12:
My apologies for the late reply. For some reason I'm not getting email about Github issues.

I'll take a look at Horizon-EDA. The two main issues are:
1. Reading library files. Getting these into a Python data structure is obviously easy since they're JSON. Hopefully, constructing the SKiDL Part objects will also be easy.
2. Outputing netlist files. I looked at the docs and scanned the schematic source file names. Nothing jumps out as to what the netlist format might be.

AppMathIsRealMath, 2020-05-13:
Some sample projects can be found in [hubble](https://github.com/carrotIndustries/hubble) and [x-band-tx](https://github.com/carrotIndustries/x-band-tx). The netlist is stored in a JSON file e.g. [top_block.json](https://github.com/carrotIndustries/x-band-tx/blob/master/top_block.json).

From a freenode chat with the creator/maintainer of horizon there's no explicit export or import of the netlist - it's pretty tightly coupled within the program. However nothing prevents editing an existing top_block.json or (possibly) creating a brand new one externally.

I think the flow with SKiDL would be to first manually create a Horizon-EDA project which would generate an empty or mostly empty top_block.json, and then generate a replacement top_block.json from SKiDL and then drive on.

So it's sort of hacky, intruding in the dataflow in a manner not explicitly designed for but apparently supported.

My thinking was to expand the range of SKiDL so perhaps we can some day crush the evil of schematics ;-) But if it's too much work for too little gain I understand. My main use of SKiDL is with Kicad and that works perfectly.

xesscorp, 2020-05-20:
Thanks for the links to the example files.

Horizon-EDA and [tEDAx](http://repo.hu/projects/tedax/) are both systems that SKiDL *could* support, but I'm currently concentrating on adding more general-purpose features. I'm time/energy constrained right now, but thanks for bringing this to my attention. I'll leave the issue open to serve as a reminder.

AppMathIsRealMath, 2020-05-23:
More general purpose features sounds good to me - thanks for having a look.

If you get time eventually and are able to expand SKiDL's range of EDA software another project is [LibrePCB](https://github.com/LibrePCB/LibrePCB) (which I know even less about than Horizon, having downloaded and played with it once about six months ago).