devbisme/skidl #84: [SKiDL BUG] Cannot run unit tests -- PermissionError

jbayless, 2020-05-22:
**Bug description**

Can't run unit tests; they all choke on the line os.remove(self.filename) in remove_log_file in logger.py, which is imported at the start of every test.

**To Reproduce**
Steps to reproduce the behavior:
1. (Running Windows 10)
2. Clone repo from source, navigate to tests folder
3. Run 'pytest' (with or without admin privileges)
4. Error output:

> 
> <<(any test file, eg. test_NC.py)>>
>    from .setup_teardown *
> setup_teardown.py: 5: in <module>
>    no_files()
> ..\skidl\skidl.py:230: in no_files
>    erc_logger.stop_file_output()
> ..\skildl\logger.py:101: in stop_file_output
>   self.removehandler(handler)
> ..\skidl\logger.py:93: in removeHandler
>    handler.remove_log_file()
> ..\skidl\logger.py:73: in remove_log_file
>    os.remove(self.filename)
> E   PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'test_NC.erc'

This above error repeats ~30 times, once for each module in the tests directory.

**Expected behavior**
Expected Pytest to run and all tests to pass.

**Desktop (please complete the following information):**
 - OS: Windows 10 (running Miniconda python)
 - SKiDL version: Tried with 0.030, both master and development version

**Additional context**
If I add a try... except(PermissionError) on the offending function, I can get the tests to run. In that case, 1 fails and 116 pass.  The test that fails is test_xspice_1, and the error is an OSError raised from _load_backend_lib indicating that ngspice.dll could not be located. (I don't think I have ngspice installed so that's not too concerning).

xesscorp, 2020-05-25:
What version of miniconda are you using?

jbayless, 2020-05-28:
I apologize for the late reply!
Ah... that seems to have been the issue. I was using 4.8.1. I updated to 4.8.3 today, and now the tests all run properly! I'll close the issue.

closed 2020-05-28 by jbayless

xesscorp, 2020-05-28:
Thanks for the update. I've never run anaconda myself. I didn't think it was the cause, I was just trying to recreate your environment so I could check the problem. Glad it's working now.