devbisme/skidl #85: Support for programmatic spice model declaration

matthuszagh, 2020-05-29:
PySpice provides the ability to programmatically declare models with the [`DeviceModel` class](https://github.com/FabriceSalvaire/PySpice/blob/910c05389ef13ea6c6b9b4cd58a55d798e7695d2/PySpice/Spice/Netlist.py#L104). For example, to include the generic BJT model at the beginning of the BJT chapter (chapter 8) in the NgSpice manual, you could do

```python
DeviceModel(name="QMOD1", modele_type="NPN level=2")
```

While there is an obvious workaround for this in skidl (write it to a file and reference it like any other spice model), this adds a layer of difficulty to using simply, generic models and adds an even greater level of difficulty for programmatic generation of models.

Would it be possible to add a skidl wrapper around this pyspice functionality? It's technically possible to specify the model from the output of `generate_netlist`, but of course at this point it's too late.

Thanks!

xesscorp, 2020-05-29:
I think this can be done in the same way I supported XSPICE models. If so, it will be a small change.

xesscorp, 2020-05-30:
I added a `DeviceModel` class to SKiDL. (It's **not** the same as the PySpice `DeviceModel` class.)

To use it, you can take the following line from the transistor amplifier example in my [PySpice tutorial notebook](https://github.com/xesscorp/skidl/blob/master/examples/spice-sim-intro/spice-sim-intro.ipynb):

```py
q = BJT(model='2n2222a')    # 2N2222A NPN transistor. The model is stored in a directory of SPICE .lib files.
```
and change it to:
```py
q = BJT(model=DeviceModel('qmod', 'npn', level=2))
```
The first argument to `DeviceModel` is an arbitrary but unique unique name for the particular device model. The second argument is a specific string for a general device model (either 'npn' or 'pnp' for bipolar transistors). The remaining keyword arguments set parameters for the model and are dependent upon the device you're modeling.

I tried this on my transistor amplifier example and it seemed to work. I'm not a big SPICE user so there may be things wrong that I don't know about. Give it a try and let me know. You can install it like this:
```terminal
pip install git+https://github.com/xesscorp/skidl/master
```

matthuszagh, 2020-05-30:
I've given this a few tests, and seems to work great!

My only thought (and I guess this applies to Xspice too) is that I wonder if there will be a bit of confusion regarding the API. Most spice models require passing in the string of the model name, wherease xspice and now generic models require passing in the object. For example, if I use something like
```python
npn_model = DeviceModel("QMOD1", "NPN", level=4, Is=1e-15)
npn = Part("pyspice", "Q", model="QMOD1")
```
I get an error. Instead, I need to do
```python
npn = Part("pyspice", "Q", model=DeviceModel("QMOD1", "NPN", level=4, Is=1e-15))
```
I personally prefer passing in the object, so I don't mind the current implementation (and at the very least think it should continue to be supported) but I could imagine this creating a bit of confusion. Just a thought.

xesscorp, 2020-05-31:
You can still use a string for the model, but in that case it's assumed to be a file name where the .model statement is stored. It doesn't refer to the name assigned in the `XspiceModel` or `DeviceModel`. In fact, maybe those names shouldn't have to be there at all since they're not used, and SKiDL could just assign some automatically-generated model names that the user would never see unless the SPICE code was viewed.

matthuszagh, 2020-06-01:
True, it might be better to auto-generate the name.