devbisme/skidl #87: [SKiDL BUG] Failed part search in zyc due to need for KISYSMOD in skidl

prazzb, 2020-07-22:
**Describe the bug**
zyc crashes due to an issue in skidl part search

**To Reproduce**
Steps to reproduce the behavior:
1. Install skidl and zyc in a python venv
2. Open zyc through terminal and search for 0805
3. See error in terminal

**Expected behavior**
load function in part_query.py reads the file fp-lib-table. That file has pointers such as  
```
${KISYSMOD}/Battery.pretty
```
which gives an error as KISYSMOD is not exported.

**Error stack trace**
```
Traceback (most recent call last):
  File "/home/prazzb/skidlEnv/lib/python3.6/site-packages/zyc/skidl_footprint_search.py", line 378, in OnSearch
    for lib_module in search_footprints_iter(search_text):
  File "/home/prazzb/skidlEnv/lib/python3.6/site-packages/skidl/part_query.py", line 287, in search_footprints_iter
    footprint_cache.load(os.path.join(path, "fp-lib-table"))
  File "/home/prazzb/skidlEnv/lib/python3.6/site-packages/skidl/part_query.py", line 251, in load
    for fn in os.listdir(uri)
FileNotFoundError: [Errno 2] No such file or directory: '${KISYSMOD}/Battery.pretty'
```

**Desktop (please complete the following information):**
 - OS - Ubuntu 18.04 LTS
 - Python version - 3.6.9
 - SKiDL version - 0.0.30 installed over pip

**Additional context**
Adding KISYSMOD in bashrc fixes this. However, doing a git grep on this project revealed that an old version removed
the need for KISYSMOD. I'm thus unclear.

xesscorp, 2020-07-22:
KISYSMOD is no longer used by SKiDL for finding the location of the part libraries (KICAD_SYMBOL_DIR is now used, instead). But KISYSMOD still needs to be defined in your environment for zyc (and PCBNEW) to resolve the directories in the fp-lib-table file.

prazzb, 2020-07-25:
I can send a pull request to the documentation. However,the doc directory is a weird mix of markdown and sphinx so I can send a patch but not properly test it.

In any case,please add the need for KISYSMOD in the documentation similar to KICAD_SYMBOL_DIR. I struggled for a while
because of the documentation not being present.

xesscorp, 2020-07-25:
Thanks, I'll take care of this.

xesscorp, 2021-07-09:
The logger will now issue a warning if undefined environment variables are found.

closed 2021-07-09 by xesscorp