devbisme/skidl #94: Part templates can connect to nets without raising any warnings/errors

voneiden, 2020-09-12:
**Describe the bug**

Currently it is possible to connect part templates to nets. In the example below, we have two test points which are connected by a template resistor. The ERC raises no issues yet the generated netlist is invalid because it references a component called "None". Yes, user error, but apparently can happen accidentally when playing around with imported objects and forgetting that one of them was a template (cough).

**To Reproduce**
Steps to reproduce the behavior:

```
from skidl import *

vcc, gnd = Net('VCC'), Net('GND')

tp_gnd = Part('Connector', 'TestPoint')
tp_vcc = Part('Connector', 'TestPoint')

tp_gnd & gnd & Part('Device', 'R', dest=TEMPLATE) & vcc & tp_vcc

ERC()
generate_netlist(file_=sys.stdout)
```

**Expected behavior**
Attempting to connect a template to a net could raise an error. ERC could also sanity check that a net is not connected to template parts. 

**Desktop (please complete the following information):**
 - OS Gentoo Linux
 - Python version 3.8.5
 - SKiDL version 0.0.30

xesscorp, 2020-09-15:
In the master branch I don't get an error:
```python
In [1]: from skidl import *                                                                                                                     

In [2]: Net() & Part('Device', 'R', dest=TEMPLATE)                                                                                              
Out[2]: [N$1: Pin None/1/~/PASSIVE, Pin None/2/~/PASSIVE]
```

In the development branch I get an error:
```python
In [2]: from skidl import *                                                                                                                     

In [3]: Net() & Part('Device','R',dest=TEMPLATE)                                                                                                
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-e0b7ac4323a1> in <module>
----> 1 Net() & Part('Device','R',dest=TEMPLATE)

/media/devb/Main/xesscorp/KiCad/tools/skidl/skidl/net.py in __and__(self, obj)
    629         from .network import Network
    630 
--> 631         return Network(self) & obj
    632 
    633     def __rand__(self, obj):

/media/devb/Main/xesscorp/KiCad/tools/skidl/skidl/network.py in __and__(self, obj)
     86         # (Use -1 index to get the output port instead of 1 because the network
     87         # may only have a single port serving as both the input and output.)
---> 88         self[-1] += ntwk[0]
     89 
     90         # Return a network consisting of the input of the first and the output of the second.

/media/devb/Main/xesscorp/KiCad/tools/skidl/skidl/net.py in connect(self, *pins_nets_buses)
    510                         ValueError,
    511                         "Can't attach a part to a net in different circuits ({}, {})!".format(
--> 512                             pn.part.circuit.name, self.circuit.name
    513                         ),
    514                     )

AttributeError: 'NoneType' object has no attribute 'name'

```
Admittedly, it's not a very descriptive error as to the exact problem, but it is an error. I'll try to make this a better experience for the user.

xesscorp, 2020-09-15:
I changed the development branch to give a more descriptive warning message.

closed 2021-02-19 by xesscorp