devbisme/skidl #103: [SKiDL BUG] KiCAD not picking up the "0-th" net

0dvictor, 2021-01-23:
**Describe the bug**
*I'm not sure whether this is a KiCAD bug or an skidl one.*

KiCAD ignores the first net when importing a netlist.
The issue appears to be that KiCAD counts from 1 but skidl counts from 0, resulting the 0-th net being ignored by KiCAD.
```
(nets
    (net (code 0) (name +1.2V)
```

KiCAD can correctly import the netlist file if manually changing the code to a non-zero number (without collision with others, of course).

**To Reproduce**
Steps to reproduce the behavior:
1. Make a circuit with multiple nets, say N;
2. Generate the netlist file;
3. Import the netlist file to KiCAD's pcbnew;
4. Check the wiring and nets;
5. => The 0-th net does not show up, i.e. only N-1 nets shows.

**Expected behavior**
All nets show up in KiCAD's pcbnew, and all wirings are correct.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS under WSL2, Windows 10 19042.782
 - Python: 3.8.5
 - SKiDL: 0.0.30
 - KiCAD: 5.99.0-8493-g91c551f4ab

**Additional context**
I'm not sure if it matters: KiCAD generated netlists seems always to have the code name quoted (e.g. `(code "1")` instead of `(code 1)`).

adamws, 2021-01-25:
checked with kicad `5.1.9` and it is working. Perhaps you should check with older kicad version

0dvictor, 2021-01-25:
> checked with kicad `5.1.9` and it is working. Perhaps you should check with older kicad version

Thanks, I'll try 5.1.9.

xesscorp, 2021-01-28:
Yesterday, I tried importing a netlist into KiCad 5.1.7 and KiCad 5.99-whatever. I didn't experience any problems with having a netlist with a code of 0.

That said, it's certainly not hard to change the starting point for numbering nets in SKiDL. Let me know if you continue to have this problem.

0dvictor, 2021-01-29:
I just tried the latest nightly (5.99.0-8694-g6983bcd55b) and I am still seeing this problem.

I think it is better for SKiDL to change the starting point. Netlists generated by KiCad's Eeschema always start from 1¹, so I'm afraid there might be little incentive for them to change the behavior back.

¹ Based on observation from 5.1.9, 5.99.0-8493-g91c551f4ab, and 5.99.0-8694-g6983bcd55b.

xesscorp, 2021-01-30:
I changed the starting point for netlist codes to 1 and pushed the change to the development branch. Let me know if it fixes the problem.

0dvictor, 2021-01-31:
Yes, the problem has been fixed. Thanks!

closed 2021-02-11 by xesscorp