devbisme/skidl #104: part ref auto-increment when a net is named is ambiguous (ex R1, L1)

NilujePerchut, 2021-02-07:
**Describe the bug**
If a net name is also a part ref  (L1, R1, R2, L2, .. (playstation stuff))
In that case, all the resistors are named R1__XX, All the the inductors are named L1_XX.

**To Reproduce**
Steps to reproduce the behaviour:
1. Instantiate a Net named R1
2. Instantiate two resistors letting skidl do the ref work
3. Resistor are named R1_1 and R1_2

**Expected behaviour**
Either:
* SKIDL can make the difference between net and part ref: resistors names=R1 and R2
* SKIDL skips the ambiguous names: resistors names=R2, R3

**Desktop (please complete the following information):**
 - OS Manjaro/GNOME
 - Python 3.9.1
 - SKiDL version 0.0.3

**Additional context**
After some digging, second expected behaviour can be achieved after the following patch:

```
diff --git a/skidl/utilities.py b/skidl/utilities.py
index f175ef3..f4d35a4 100644
--- a/skidl/utilities.py
+++ b/skidl/utilities.py
@@ -367,7 +367,7 @@ def get_unique_name(lst, attrib, prefix, initial=None):
 
     # If the initial name is None, then create a name based on the prefix
     # and the smallest unused number that's available for that prefix.
-    if not name:
+    if not initial:
 
         # Do a binary search for a unique name formed from the prefix + number.
         n = 1  # Starting number to append to the prefix.
```

xesscorp, 2021-02-11:
Thanks for your bug report! SKiDL wasn't distinguishing between net names and part names (everything was kept in the same name list). I fixed this in the development branch.

NilujePerchut, 2021-02-11:
Thanks for the quick fix !

closed 2021-02-11 by NilujePerchut