devbisme/skidl #107: [SKiDL BUG] Copying template part

adamws, 2021-03-04:
**Describe the bug**
Copying of template part produces one part extra. See following example:

**To Reproduce**
```
>>> from skidl import *
>>> r_template = Part("Device", 'R', dest=TEMPLATE)
>>> r1, r2 = 2 * r_template(value="10k")
>>> generate_netlist(file_=sys.stdout)
ERROR: No footprint for R/R1.
ERROR: No footprint for R/R2.
ERROR: No footprint for R/R3.

0 warnings found during netlist generation.
3 errors found during netlist generation.

(export (version D)
  (design
    (source "/home/aws/.local/lib/python3.9/site-packages/skidl/Circuit.py")
    (date "03/05/2021 12:05 AM")
    (tool "SKiDL (0.0.30)"))
  (components
    (comp (ref R1)
      (value 10k)
      (footprint "No Footprint")
      (fields
        (field (name F0) R)
        (field (name F1) R))
      (libsource (lib Device) (part R))
      (sheetpath (names /top/13040407304974508527) (tstamps /top/13040407304974508527)))
    (comp (ref R2)
      (value 10k)
      (footprint "No Footprint")
      (fields
        (field (name F0) R)
        (field (name F1) R))
      (libsource (lib Device) (part R))
      (sheetpath (names /top/4893829896871865379) (tstamps /top/4893829896871865379)))
    (comp (ref R3)
      (value 10k)
      (footprint "No Footprint")
      (fields
        (field (name F0) R)
        (field (name F1) R))
      (libsource (lib Device) (part R))
      (sheetpath (names /top/5370436225866403464) (tstamps /top/5370436225866403464))))
  (nets)
)
```

**Expected behavior**

One line approach which is presented in documentation is working fine

```
r1, r2 = 2 * Part("Device", 'R', dest=TEMPLATE) 
```

but in my use case I wanted to create single template for resistor for selected footprint and use it to instantiate groups of parts with same value, like this: 

```
R = Part("Device", "R", TEMPLATE, footprint="Resistor_SMD:R_0603_1608Metric")
r1, r2 = 2 * R(value="10k")
r3, r4 = 2 * R(value="20k")
```
which would save me typing, but unfortunately doesn't work as expected.

**Desktop (please complete the following information):**
 - on linux
 - Python version `Python 3.9.1`
 - SKiDL version `0.0.30`

xesscorp, 2021-03-05:
This problem is caused by the way Python operates. I don't currently have a way to solve it.

Here is the sequence of operations that happens with `r1, r2 = 2 * Part("Device", "R", dest=TEMPLATE)`:
1. A template resistor is made.
2. The `*` operator makes two real copies of the template.
3. The two real copies are assigned to `r1` and `r2`.

Here is the sequence of operations that happens with `r1, r2 = 2 * r_template(value="10K")`:
1. The resistor template is called with a value of 10K. This causes it to create a real resistor (`R1`) with that value of resistance.
2. The `*` operator is applied to that real resistor, creating two more real 10K resistors (`R2` and `R3`). Now there are three real resistors.
3. The two new resistors are assigned to `r1` and `r2`.

Note that this would work if you wrote `r1, r2 = 2 * r_template`:
1. The `*` operator would make two real copies from the template. (Note that the template was not called; it just had the `*` operator applied to it.)
2. The two real copies are assigned to `r1` and `r2`. However, then you would have to set the resistor values separately (more verbose).

The way to get what you want is this: `r1, r2 = r_template(num_copies=2, value="10K")`. This calls the resistor template and tells it to make two real resistors with a value of 10K.

I'll admit that the use of the `*` operator the way you used it is more attractive and clear. But I struggled with that a few years ago and couldn't figure out how to do it. Maybe you have an idea.

adamws, 2021-03-05:
thank you for detailed explanation. `r1, r2 = r_template(num_copies=2, value="10K")` is almost as verbose as `*` operator and it's perfectly fine for me. I'm not an python expert to suggest something better.

I would just suggest to mention this in [docs](https://xess.com/skidl/docs/_site/index.html) since `num_copies` option is currently not described.

mtnrbq, 2021-04-11:
@adamws 

I did this to work around the issue:
```python
def mct(template:Part, value, copies=1):
    return template(num_copies=copies,value = value)

def mra(value, copies=1):
    template=Part(lib='Device.lib', name='R', dest=TEMPLATE, footprint='Resistor_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal') 
    return mct(template,value,copies)
```
usage:
```python
r1,r2,r3,r4 =mra('10k',4)  # 4 x 10k Ohm axial resistors
r5=mra('100k')             # 1 x 100k Ohm axial resistor
r6=mra('100')              # 1 x 10 Ohm axial resistor
r7,r8=mra('1k',2)          # 2 x 1K Ohm axial resistor
```

closed 2021-07-09 by xesscorp