devbisme/skidl #109: zyc on ubuntu

eljuligallego, 2021-04-02:
Hello,
how I have to launch zyc on ubuntu?. in one video I saw just write zyc in command line in windows will work but after install the skidl via github with commad line like proposed in zyc help [webpage](https://xess.com/skidl/docs/_site/blog/worst-part-of-skidl) the zyc command is not recognized and skidl is working perfect.
Maybe some path not configured?.

Thanks

xesscorp, 2021-04-02:
Did you do `pip install zyc`?

eljuligallego, 2021-04-02:
Sorry, from webpage mention before I understand zyc was included with skidl since no mention to pip intall zyc command in install instructions.

But, just run sudo pip install zyc (had to run sudo apt-get install libgtk-3-dev before) but now it freeze when installing the wxpython...(cc1plus process consumes all cpu but never ends).

Thanks for reply.

eljuligallego, 2021-04-03:
Done, thanks.

closed 2021-04-03 by eljuligallego