devbisme/skidl #114: [SKiDL BUG]

2themaxx, 2021-05-13:
**Describe the bug**
Skidl 1.0 is incompatible with PySpice 1.4.3

**To Reproduce**
Use conda to make a new virtual environment with PySpice 1.4.3
Use Pip to install skidl 1.0.0
Run example that loads library NCP1117 (section in spice-sim-example).

Result:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-15-93a836f6504a> in <module>
      4 
      5 vin = V(ref='VIN', dc_value=8@u_V)      # Input power supply.
----> 6 vreg = Part('NCP1117', 'ncp1117_33-x')  # Voltage regulator from ON Semi part lib.
      7 print(vreg)                             # Print vreg pin names.
      8 r = R(value=470 @ u_Ohm)                # Load resistor on regulator output.

~/miniconda3/envs/circuitsimpip/lib/python3.8/site-packages/skidl/part.py in __init__(self, lib, name, dest, tool, connections, part_defn, circuit, **kwargs)
    186                 libname = lib
    187                 try:
--> 188                     lib = SchLib(filename=libname, tool=tool)
    189                 except FileNotFoundError as e:
    190                     if skidl.QUERY_BACKUP_LIB:

~/miniconda3/envs/circuitsimpip/lib/python3.8/site-packages/skidl/schlib.py in __init__(self, filename, tool, lib_section, **attribs)
     98                 )
     99             else:
--> 100                 load_func(
    101                     filename, skidl.lib_search_paths[tool], lib_section=lib_section
    102                 )

~/miniconda3/envs/circuitsimpip/lib/python3.8/site-packages/skidl/tools/spice.py in _load_sch_lib_(self, filename, lib_search_paths_, lib_section)
    115 
    116     # Read the Spice library from the given path.
--> 117     spice_lib = SpiceLibrary(
    118         root_path=spice_lib_path, recurse=True, section=lib_section
    119     )

TypeError: __init__() got an unexpected keyword argument 'recurse'

```
**Expected behavior**
Run without errors and graph input vs output

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Python version: 3.8
 - SKiDL version: 1.0.0

**Additional context**
Add any other context about the problem here.

2themaxx, 2021-05-13:
### Workaround in the mean time:

1. Prep virtual environment with everything but PySpice and skidl
2. pip install git+https://github.com/FabriceSalvaire/PySpice
3. pip install skidl
4. example works as expected

xesscorp, 2021-05-13:
I made some changes to PySpice to support the SKY130 PDK and submitted a PR to Fabrice last year. These changes also made SKiDL incompatible with PySpice 1.4.3 but I didn't notice that since I still had my modified PySpice installed locally.

Fabrice has recently incorporated my PR into PySpice. That's why using his repo solves this problem. I believe he is getting ready to release the next version of PySpice on PyPi and that will completely solve the problem.

xesscorp, 2021-07-08:
Fabrice released PySpice 1.5 on May 15th, so this problem shouldn't appear again.

closed 2021-07-08 by xesscorp