devbisme/skidl #118: [SKiDL BUG]

Dunkelwind, 2021-07-07:
skin.svg  file has no correct format for netlistsvg.

netlistsvg error:
```
/usr/local/lib/node_modules/netlistsvg/built/Skin.js:11
                return e[1]['s:pid'].startsWith(prefix);
                                     ^

TypeError: Cannot read property 'startsWith' of undefined
    at /usr/local/lib/node_modules/netlistsvg/built/Skin.js:11:38
    at arrayFilter (/usr/local/lib/node_modules/netlistsvg/node_modules/lodash/lodash.js:596:11)
    at Function.filter (/usr/local/lib/node_modules/netlistsvg/node_modules/lodash/lodash.js:9241:14)
    at Object.getPortsWithPrefix (/usr/local/lib/node_modules/netlistsvg/built/Skin.js:9:23)
    at Cell.buildElkChild (/usr/local/lib/node_modules/netlistsvg/built/Cell.js:213:36)
    at /usr/local/lib/node_modules/netlistsvg/built/elkGraph.js:12:18
    at Array.map (<anonymous>)
    at Object.buildElkGraph (/usr/local/lib/node_modules/netlistsvg/built/elkGraph.js:11:33)
    at Object.render (/usr/local/lib/node_modules/netlistsvg/built/index.js:43:29)
    at render (/usr/local/lib/node_modules/netlistsvg/bin/netlistsvg.js:23:9)
    at parseFiles (/usr/local/lib/node_modules/netlistsvg/bin/netlistsvg.js:57:9)
    at fs.readFile (/usr/local/lib/node_modules/netlistsvg/bin/netlistsvg.js:41:17)
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:511:3)
```
I think **s:pid** should be in the same root <g> tag
i.e.
```xml
<g s:type="PMOS_GSD_1_" s:width="112.5" s:height="132.0" transform="translate(60.0,72.0)">
			<s:alias val="PMOS_GSD_1_" />
			<path d="M 27.0 0.0 L 15.0 -4.5 L 15.0 4.5 L 27.0 0.0" style="stroke-width:1.0"
				class="$cell_id symbol pen_fill" />
			<circle cx="19.5" cy="0.0" r="33.0" style="stroke-width:3.0" class="$cell_id symbol " />
			<path d="M 3.0 0.0 L -30.0 0.0" style="stroke-width:1.0" class="$cell_id symbol " />
			<path d="M 3.0 -22.5 L 3.0 22.5" style="stroke-width:3.0" class="$cell_id symbol " />
			<path d="M 9.0 21.0 L 30.0 21.0" style="stroke-width:1.0" class="$cell_id symbol " />
			<path d="M 9.0 15.0 L 9.0 27.0" style="stroke-width:3.0" class="$cell_id symbol " />
			<path d="M 9.0 -6.0 L 9.0 6.0" style="stroke-width:3.0" class="$cell_id symbol " />
			<path d="M 9.0 -21.0 L 30.0 -21.0" style="stroke-width:1.0" class="$cell_id symbol " />
			<path d="M 9.0 -27.0 L 9.0 -15.0" style="stroke-width:3.0" class="$cell_id symbol " />
			<path d="M 30.0 -30.0 L 30.0 -21.0" style="stroke-width:1.0" class="$cell_id symbol " />
			<path d="M 30.0 30.0 L 30.0 0.0 L 9.0 0.0" style="stroke-width:1.0" class="$cell_id symbol " />
			<path d="M -60.0 0.0 L -30.0 0.0" style="stroke-width:1.0" class="$cell_id symbol" />
			<text class='pin_num_text' text-anchor='end' x='-30.0' y='0.0'
				transform='rotate(0 -30.0 0.0) translate(-6.0 -2.25)' style='font-size:15.0px'> 1 </text>
			<path d="M 30.0 60.0 L 30.0 30.0" style="stroke-width:1.0" class="$cell_id symbol" />
			<text class='pin_num_text' text-anchor='end' x='30.0' y='30.0'
				transform='rotate(-90 30.0 30.0) translate(-6.0 -2.25)' style='font-size:15.0px'> 2 </text>
			<path d="M 30.0 -72.0 L 30.0 -30.0" style="stroke-width:1.0" class="$cell_id symbol" />
			<text class='pin_num_text' text-anchor='start' x='30.0' y='-30.0'
				transform='rotate(-90 30.0 -30.0) translate(6.0 -2.25)' style='font-size:15.0px'> 3 </text>
			<text class='part_ref_text' text-anchor='start' x='60.0' y='-15.0'
				transform='rotate(0 60.0 -15.0) translate(0.0 7.5)' style='font-size:15.0px' s:attribute="ref"> X
			</text>
			<text class='part_name_text' text-anchor='start' x='60.0' y='15.0'
				transform='rotate(0 60.0 15.0) translate(0.0 7.5)' style='font-size:15.0px' s:attribute="value"> X
			</text>
			<g s:x="0.0" s:y="0" s:pid="1" s:position="left"/>
			<g s:x="30.0" s:y="60" s:pid="2" s:position="bottom"/>
			<g s:x="30.0" s:y="-72" s:pid="3" s:position="top"/>
	</g>
```
not working because extra <g transform...:
```xml
<g s:type="PMOS_GSD_1_" s:width="112.5" s:height="132.0" >
<s:alias val="PMOS_GSD_1_"/>
<g transform="translate(60.0,72.0)">
<path d="M 27.0 0.0 L 15.0 -4.5 L 15.0 4.5 L 27.0 0.0" style="stroke-width:1.0" class="$cell_id symbol pen_fill" />
<circle cx="19.5" cy="0.0" r="33.0" style="stroke-width:3.0" class="$cell_id symbol " />
<path d="M 3.0 0.0 L -30.0 0.0" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M 3.0 -22.5 L 3.0 22.5" style="stroke-width:3.0" class="$cell_id symbol " />
<path d="M 9.0 21.0 L 30.0 21.0" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M 9.0 15.0 L 9.0 27.0" style="stroke-width:3.0" class="$cell_id symbol " />
<path d="M 9.0 -6.0 L 9.0 6.0" style="stroke-width:3.0" class="$cell_id symbol " />
<path d="M 9.0 -21.0 L 30.0 -21.0" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M 9.0 -27.0 L 9.0 -15.0" style="stroke-width:3.0" class="$cell_id symbol " />
<path d="M 30.0 -30.0 L 30.0 -21.0" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M 30.0 30.0 L 30.0 0.0 L 9.0 0.0" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M -60.0 0.0 L -30.0 0.0" style="stroke-width:1.0" class="$cell_id symbol"/>
<text class='pin_num_text' text-anchor='end' x='-30.0' y='0.0' transform='rotate(0 -30.0 0.0) translate(-6.0 -2.25)' style='font-size:15.0px'  > 1 </text>
<path d="M 30.0 60.0 L 30.0 30.0" style="stroke-width:1.0" class="$cell_id symbol"/>
<text class='pin_num_text' text-anchor='end' x='30.0' y='30.0' transform='rotate(-90 30.0 30.0) translate(-6.0 -2.25)' style='font-size:15.0px'  > 2 </text>
<path d="M 30.0 -72.0 L 30.0 -30.0" style="stroke-width:1.0" class="$cell_id symbol"/>
<text class='pin_num_text' text-anchor='start' x='30.0' y='-30.0' transform='rotate(-90 30.0 -30.0) translate(6.0 -2.25)' style='font-size:15.0px'  > 3 </text>
<text class='part_ref_text' text-anchor='start' x='60.0' y='-15.0' transform='rotate(0 60.0 -15.0) translate(0.0 7.5)' style='font-size:15.0px' s:attribute="ref" > X </text>
<text class='part_name_text' text-anchor='start' x='60.0' y='15.0' transform='rotate(0 60.0 15.0) translate(0.0 7.5)' style='font-size:15.0px' s:attribute="value" > X </text>
</g>
<g s:x="0.0" s:y="72.0" s:pid="1" s:position="left"/>
<g s:x="90.0" s:y="132.0" s:pid="2" s:position="bottom"/>
<g s:x="90.0" s:y="0.0" s:pid="3" s:position="top"/>
</g>
```

xesscorp, 2021-07-08:
Please check and make sure you're using the most recent version of netlistsvg (version 1.0.2 on my system). This error looks similar to problems I encountered when developing the SVG code for SKiDL and I made a PR to netlistsvg to solve it. You can download the latest code from the repo with:
```bash
npm install github:nturley/netlistsvg
```

Dunkelwind, 2021-07-08:
I have already tested with V1.0.2
Now, I have installed the github-version:
```
$ npm ls 
dunkelwind@ /home/dunkelwind
└── netlistsvg@1.0.2 (git+ssh://git@github.com/nturley/netlistsvg.git#d083a3c16eb94a5aecf5e5d4d615fab007a426a2)
```
same error:
```
/usr/local/bin/netlistsvg main.json --skin main_skin.svg 
/usr/local/lib/node_modules/netlistsvg/built/Skin.js:11
                return e[1]['s:pid'].startsWith(prefix);
                                     ^

TypeError: Cannot read property 'startsWith' of undefined
    at /usr/local/lib/node_modules/netlistsvg/built/Skin.js:11:38
    at arrayFilter (/usr/local/lib/node_modules/netlistsvg/node_modules/lodash/lodash.js:596:11)
    at Function.filter (/usr/local/lib/node_modules/netlistsvg/node_modules/lodash/lodash.js:9241:14)
    at Object.getPortsWithPrefix (/usr/local/lib/node_modules/netlistsvg/built/Skin.js:9:23)
    at Cell.buildElkChild (/usr/local/lib/node_modules/netlistsvg/built/Cell.js:213:36)
    at /usr/local/lib/node_modules/netlistsvg/built/elkGraph.js:12:18
    at Array.map (<anonymous>)
    at Object.buildElkGraph (/usr/local/lib/node_modules/netlistsvg/built/elkGraph.js:11:33)
    at Object.render (/usr/local/lib/node_modules/netlistsvg/built/index.js:43:29)
    at render (/usr/local/lib/node_modules/netlistsvg/bin/netlistsvg.js:23:9)
    at /usr/local/lib/node_modules/netlistsvg/bin/netlistsvg.js:57:9
    at /usr/local/lib/node_modules/netlistsvg/bin/netlistsvg.js:41:17

```
svg:

``` xml
<svg xmlns="http://www.w3.org/2000/svg"     xmlns:xlink="http://www.w3.org/1999/xlink"     xmlns:s="https://github.com/nturley/netlistsvg">  <s:properties    constants="false"    splitsAndJoins="false"    genericsLaterals="true">    <s:layoutEngine        org.eclipse.elk.layered.spacing.nodeNodeBetweenLayers="5"        org.eclipse.elk.layered.compaction.postCompaction.strategy="4"        org.eclipse.elk.spacing.nodeNode= "50"        org.eclipse.elk.direction="DOWN"/>  </s:properties><style>svg {  stroke: #000;  fill: none;  stroke-linejoin: round;  stroke-linecap: round;}text {  fill: #000;  stroke: none;  font-size: 10px;  font-weight: bold;  font-family: "Courier New", monospace;}.skidl_text {  fill: #999;  stroke: none;  font-weight: bold;  font-family: consolas, "Courier New", monospace;}.pin_num_text {    fill: #840000;}.pin_name_text {    fill: #008484;}.net_name_text {    font-style: italic;    fill: #840084;}.part_text {    fill: #840000;}.part_ref_text {    fill: #008484;}.part_name_text {    fill: #008484;}.pen_fill {    fill: #840000;}.background_fill {    fill: #FFFFC2}.nodelabel {  text-anchor: middle;}.inputPortLabel {  text-anchor: end;}.splitjoinBody {  fill: #000;}.symbol {  stroke-linejoin: round;  stroke-linecap: round;  stroke: #840000;}.detail {  stroke-linejoin: round;  stroke-linecap: round;  fill: #000;}</style><!-- signal --><g s:type="inputExt" s:width="30" s:height="20" transform="translate(0,0)">  <text x="-2" y="12" text-anchor='end' class="$cell_id pin_name_text" s:attribute="ref">input</text>  <s:alias val="$_inputExt_"/>  <path d="M0,0 V20 H15 L30,10 15,0 Z" class="$cell_id symbol"/>  <g s:x="30" s:y="10" s:pid="Y" s:position="right"/></g><g s:type="outputExt" s:width="30" s:height="20" transform="translate(0,0)">  <text x="32" y="12" class="$cell_id pin_name_text" s:attribute="ref">output</text>  <s:alias val="$_outputExt_"/>  <path d="M30,0 V20 H15 L0,10 15,0 Z" class="$cell_id symbol"/>  <g s:x="0" s:y="10" s:pid="A" s:position="left"/></g><!-- signal --><!-- builtin --><g s:type="generic" s:width="30" s:height="40" transform="translate(0,0)">  <text x="15" y="-4" class="nodelabel $cell_id" s:attribute="ref">generic</text>  <rect width="30" height="40" x="0" y="0" s:generic="body" class="$cell_id"/>  <g transform="translate(30,10)"     s:x="30" s:y="10" s:pid="out0" s:position="right">    <text x="5" y="-4" class="$cell_id">out0</text>  </g>  <g transform="translate(30,30)"     s:x="30" s:y="30" s:pid="out1" s:position="right">    <text x="5" y="-4" class="$cell_id">out1</text>  </g>  <g transform="translate(0,10)"     s:x="0" s:y="10" s:pid="in0" s:position="left">      <text x="-3" y="-4" class="inputPortLabel $cell_id">in0</text>  </g>  <g transform="translate(0,30)"     s:x="0" s:y="30" s:pid="in1" s:position="left">    <text x="-3" y="-4" class="inputPortLabel $cell_id">in1</text>  </g></g><!-- builtin -->
<g s:type="MPMOS_1_HR" s:width="144.0" s:height="132.0" >
<s:alias val="MPMOS_1_HR"/>
<g transform="translate(72.0,60.0)">
<path d="M 0.0 -60.0 L 7.5 -37.5 L -7.5 -37.5 L 0.0 -60.0" style="stroke-width:1.0" class="$cell_id symbol pen_fill" />
<path d="M 30.0 15.0 L -30.0 15.0" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M -30.0 7.5 L -30.0 -30.0" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M 30.0 -30.0 L 30.0 7.5" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M 0.0 -60.0 L 0.0 7.5" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M -30.0 7.5 L 30.0 7.5" style="stroke-width:1.0" class="$cell_id symbol " />
<path d="M -72.0 -30.0 L -30.0 -30.0" style="stroke-width:1.0" class="$cell_id symbol"/>
<text class='pin_num_text' text-anchor='end' x='-30.0' y='-30.0' transform='rotate(0 -30.0 -30.0) translate(-6.0 -2.25)' style='font-size:15.0px'  > 1 </text>
<text class='pin_name_text' text-anchor='start' x='-30.0' y='-30.0' transform='rotate(0 -30.0 -30.0) translate(6.0 3.0)' style='font-size:15.0px'  > D </text>
[... cut ...]
<path d="M 9.0 21.0 L 30.0 21.0" style="stroke-width:1.0" class="$cell_id symbol " />
[... 50 lines cut ...]
        },
        "Q1": {
          "attributes": {
            "value": "PMOS_GSD"
          },
          "connections": {
            "3": [
              2
            ]
          },
          "port_directions": {
            "1": "input",
            "2": "input",
            "3": "input"
          },
          "type": "PMOS_GSD_1_"
        }
      },
      "ports": {}
    }
  }
}
```

I'm at a loss.

Dunkelwind, 2021-07-08:
above:
skin-svg
netlist-json

python:

``` python
from skidl import *

mosfet = Part("pspice","MPMOS")
mosfet.symtx = "HR"
pmos = Part("DeviceSteffen","PMOS_GSD")
n01=Net("n01")
mosfet[1] += mosfet[2]
n01 += mosfet[3]
pmos[3] += mosfet[3]

generate_svg()

```

just test
```
netlistsvg main.json --skin main_skin.svg 
```
on your system.

Dunkelwind, 2021-07-09:
A work around:
``` diff
--- /home/dunkelwind/Schreibtisch/kicad.py	2021-07-06 06:14:53.036493000 +0200
+++ /home/dunkelwind/.local/lib/python3.6/site-packages/skidl/tools/kicad.py	2021-07-09 10:41:47.357593639 +0200
@@ -1603,24 +1603,25 @@
                     's:type="{symbol_name}"',
                     's:width="{bbox.w}"',
                     's:height="{bbox.h}"',
-                    ">",
+ #                   ">",
                 ]
             ).format(**locals())
         )
 
-        # Add part alias.
-        svg.append('<s:alias val="{symbol_name}"/>'.format(**locals()))
-
         # Group text & graphics and translate so bbox.min is at (0,0).
         translate = bbox.min * -1
         svg.append(
-            '<g transform="translate({translate.x},{translate.y})">'.format(**locals())
+            ' transform="translate({translate.x},{translate.y})">'.format(**locals())
         )
+
+        # Add part alias.
+        svg.append('<s:alias val="{symbol_name}"/>'.format(**locals()))
+
         # Add part unit text and graphics.
         svg.extend(unit_filled_svg[unit])  # Filled items go on the bottom.
         svg.extend(unit_svg[unit])  # Then unfilled items.
         svg.extend(unit_txt_svg[unit])  # Text comes last.
-        svg.append("</g>")
+    #    svg.append("</g>")
 
         # Place a visible bounding-box around symbol for trouble-shooting.
         show_bbox = False
@@ -1643,7 +1644,7 @@
         # Keep the pins out of the grouped text & graphics but adjust their coords
         # to account for moving the bbox.
         for pin_info in unit_pin_info[unit]:
-            pin_pt = Point(pin_info.x, pin_info.y) + translate
+            pin_pt = Point(pin_info.x, pin_info.y)# + translate
             side = pin_info.side
             pid = pin_info.pid
             pin_svg = '<g s:x="{pin_pt.x}" s:y="{pin_pt.y}" s:pid="{pid}" s:position="{side}"/>'.format(

```

But I don't know if that's the right way...

xesscorp, 2021-07-11:
I had a real hard time getting your example (or any example) to work because of my installation of `node` and `npm`. An old version of `netlistsvg` is installed when you use `npm install netlistsvg`. When you use `npm install github:nturley/netlistsvg` to install the newest version locally, `skidl` still defaults to using the global (but out-of-date) version of `netlistsvg`. When I tried to update the global version with `npm install github:nturley/netlistsvg -g`, Ubuntu wouldn't allow it because of directory permissions (even when using `sudo`).

So I completely removed `node` and `npm` and reinstalled the latest versions. Then all my examples began working as well as this version of your example:
```python
from skidl import *

mosfet = Part("pspice","MPMOS")
mosfet.symtx = "HL"  # Had a problem using "R" rotations.
#pmos = Part("DeviceSteffen","PMOS_GSD")  # I don't have this library.
pmos = Part("pspice","MPMOS")  # I substituted this part.
n01=Net("n01")
mosfet[1] += mosfet[2]
n01 += mosfet[3]
pmos[3] += mosfet[3]

generate_svg()
```
So I think you shouldn't need your work-around if you get the latest version of `netlistsvg` installed correctly.

One problem did arise when the second PMOS transistor in your example was rotated to the right: the `elkjs` module threw an exception. The exception didn't happen if I rotated the transistor to the left or applied any other type of transformation. I haven't yet found the cause of this error.

If you think your problems might also be caused by an out-of-date installation of `node`, `npm`, or `netlistsvg`, then here are the steps I used to re-install on Ubuntu:
```bash
cd ~
sudo apt remove node npm nodejs

sudo rm -rf /usr/local/bin/npm 
sudo rm -rf /usr/local/share/man/man1/node* 
sudo rm -rf /usr/local/lib/dtrace/node.d
rm -rf ~/.npm
rm -rf ~/.node-gyp
sudo rm -rf /opt/local/bin/node
sudo rm -rf /opt/local/include/node
sudo rm -rf /opt/local/lib/node_modules
sudo rm -rf /usr/local/lib/node*
sudo rm -rf /usr/local/include/node*
sudo rm -rf /usr/local/bin/node*

curl -sL https://raw.githubusercontent.com/nvm-sh/nvm/v0.35.0/install.sh -o install_nvm.sh
bash install_nvm.sh

nvm install stable
nvm use stable
npm install nturley/netlistsvg
```
If you're sure you have the newest version of `netlistsvg`, then we'll have to keep looking for a solution. But I got exactly the same error running your example as you did when I was using an older version of `netlistsvg`.

Dunkelwind, 2021-07-12:
Thank you for your effort.
You are right, it was the wrong version. 
It took me several hours to fix the problem :(
The current version of netlistsvg works with the work-around too.
Even the rotated transistor.
Without the work-around, I can confirm these problem (right rotated).

devbisme, 2021-09-09:
I finally got back to this and verified that your fix works for all my test cases and also solves the problem with the rotated transistor. I've checked the changes into the development branch. Thanks!

closed 2021-09-09 by devbisme