devbisme/skidl #124: [SKiDL BUG] symbol ends with _0_0

shanzez, 2021-07-13:
**Describe the bug**
Did you mean to filter out '2'?

Symbols that end with _0_0 are being filtered out (conversion flag of '0') and causing execution to fail like:
```
Traceback (most recent call last):
  File "/Users/shannonkerr/temp/gen2/boards/main.py", line 139, in <module>
    usb_connectors['MT1', 'MT2', 'MT3', 'MT4'] += gnd # todo - connect to chassis?
TypeError: unsupported operand type(s) for +=: 'NoneType' and 'Net'
```
Conversion flag:
_1 means use 1st
_2 means use 2nd
_0 means use either

Many KiCad symbols use this and cannot be imported because of this.

**To Reproduce**
Steps to reproduce the behavior:

```
skidl.Part('CM4IO', 'USB_67298-4090', footprint='todo')
```

**Expected behavior**
No errors during execution.

**Desktop (please complete the following information):**
 - Debian Linux Buster
 - Python version: 3.9
 - SKiDL version: 1.0.0

**Additional context**
Comment out these lines (854 and 855) in kicad.py and you can execute successfully:
```
 853         # Don't add this unit to the part if the conversion flag is 0.
 854         #if not conversion_flag:
 855         #    continue
```

devbisme, 2021-07-15:
I'm not sure what this means:
> Did you mean to filter out '2'?

I don't have any official info on V6 schematic libs except for the two references in my code. The [V6 library format doc](https://docs.google.com/document/d/1lyL_8FWZRouMkwqLiIt84rd2Htg4v1vz8_2MzRKHRkc/edit) doesn't really mention these suffixes appended to the symbol-unit names. And the [source code](https://gitlab.com/kicad/code/kicad/-/blob/master/eeschema/sch_plugins/kicad/sch_sexpr_parser.cpp) is nearly devoid of comments. So I made most of my decisions based on just looking through the V6 symbol libraries. If you have more solid or detailed information, I would like some pointers to it.

shanzez, 2021-07-20:
Sorry about that.

> Did you mean to filter out '2'?

Did you mean to filter out conversion flag 2?

shanemmattner, 2022-10-07:
I ran into this issue as well when I imported a part to a custom library.  Thanks @shanzez for figuring out what's going on.  After changing `AD8137YRZ_0_0` to `AD8137YRZ_0_1` the new part worked fine.  Here is the non-working `kicad_symbol` file (renamed to `.txt` so I can upload it)
[AD8137YRZ.txt](https://github.com/devbisme/skidl/files/9730184/AD8137YRZ.txt)
[AD8137YRZ_working.txt](https://github.com/devbisme/skidl/files/9730194/AD8137YRZ_working.txt)

devbisme, 2022-10-09:
I'll look into it.

parall4x, 2022-10-15:
I ran into a problem of which this is the root cause. I think the last number in `NAME_x_x` should not be regarded as a conversion flag, but as a sub unit number. 
I've noticed that some part definitions in kicad symbol files don't store pin information in this section, but for instance `74xx.kicad_sym` does.

See part `74LS08` for instance, this has section `74LS08_5_0` holding information about the power pins.

I've created a PR that should fix the issue, but please review.

commit be1188df (2022-11-09): Should fix #124
closed 2022-11-09 by devbisme