devbisme/skidl #127: [SKiDL BUG] generate_pcb()

r4dr3fr4d, 2021-09-03:
If I add `generate_pcb()` to the end of the SKIDL example to try to something I can feed into the pcbnew module, I get the following.  Seems to be kinet2pcb....is fp_libs supposed to contain the full names of the Part footprints?

```
No errors or warnings found during netlist generation.

Traceback (most recent call last):
  File "/home/rayce/linux_bkp/pcb_experiment.py", line 69, in <module>
    generate_pcb()
  File "/usr/lib/python3.9/site-packages/skidl-1.1.0-py3.9.egg/skidl/circuit.py", line 607, in generate_pcb
    gen_func(file_)  # Generate the PCB file from the netlist.
  File "/usr/lib/python3.9/site-packages/skidl-1.1.0-py3.9.egg/skidl/tools/kicad.py", line 1014, in _gen_pcb_
    kinet2pcb.kinet2pcb(self, file_)
  File "/usr/lib/python3.9/site-packages/kinet2pcb-0.1.4-py3.9.egg/kinet2pcb/kinet2pcb.py", line 154, in kinet2pcb
    lib_uri = fp_libs[fp_lib]
KeyError: 'Resistor_SMD.pretty'
```

r4dr3fr4d, 2021-09-03:
I'm wanting to try to keep the PCB design process in Python.  I saw [this exchange](https://forum.kicad.info/t/loading-netlist-with-python-script/19507) and thought generate_pcb() might be a way to create a pcbnew board object by running pcbnew.LoadBoard() on the .kicad_pcb.  Is this sound, or is there another way I should be doing this?

devbisme, 2021-09-05:
`fp_libs` is a `dict` that holds the full path to each file containing footprints. The key for each dictionary entry should be the nickname for the library file. I think the right key should be `Resistor_SMD` without the `.pretty` suffix.

The `.kicad_pcb` file output by `generate_pcb` should be acceptable to `pcbnew.LoadBoard` although I haven't tried it. I have been able to load it using `PCBNEW`.

r4dr3fr4d, 2021-09-05:
Hmm, fp_libs (loaded from LibURIs())  seems to be an empty dictionary....I added a print to verify at line 135, right before the backtrace occurs.   Could I be missing a package or something I wonder?

Yes, I'm already modelling a custom keyboard in just Python using cadquery and would like to extend that to the PCB design rather than learning a GUI, if possible.  I'm a little surprised it's not a common workflow yet, maybe in the not-so-distant future?  

I'm wanting to create a little single-switch PCB I can use many of, like [this one](https://keeb.io/products/amoeba-single-switch-pcbs) - in fact, I'm trying to open that amoeba's .kicad_pcb file and just edit the dimensions to fit my 3d model.