devbisme/skidl #131: A more human readable syntax is possible

ceremcem, 2021-09-18:
I've came across with this project recently and I'm glad that there is another project aiming programmatic design. 

With all due respect, current syntax seems too complex for a human eye, at least IMHO. A more readable syntax is possible. Here is the equivalent circuit in the README: 

```ls
my-circuit = 
    iface: "A B OUT Vcc DGND"
    bom:
        Q_PNP_CBE: "q1 q2"
        R:
            10K: "r1 r2 r3 r4 r5"
    netlist:
        Vcc: "q1.e q2.e"
        DGND: "r5.2 r3.2"
        1: "r4.1 r3.1 q1.e"
        2: "r4.2 q2.b"
        OUT: "r5.1 q2.c"
        3: "q1.b r2.2 r1.2"
        A: "r1.1"
        B: "r2.1"
```

This is a simple JSON object (actually, LSON object: Livescript JSON) which you can define every kind of schematic. You can find additional details here: https://github.com/aktos-io/aecad/blob/master/docs/creating-circuit.md

Best

devbisme, 2021-09-20:
Thanks for your input! I would urge you to post this on the SKiDL discussions thread so your ideas and aeCAD get exposed to a wider audience.

ceremcem, 2021-09-20:
Closing regarding to https://github.com/xesscorp/skidl/discussions/133.

closed 2021-09-20 by ceremcem