devbisme/skidl #135: How to code a new part and footprint?

r4dr3fr4d, 2021-10-04:
I'm wanting to create a simple pcb containing a very simple part - a cherry mx switch, which has only two pins (and some mounting holes).  I'm mucking around with custom kicad libraries/modules and trying to search from within those since the basic kicad library doesn't seem to include it, but I'm also interested in defining the part by just using skidl and bypassing the need for kicad entirely.  

Just messing around, I try to make a child class of Part:

```
class mx(Part):
    def __init__(self):
        pr = Pin(num=1, name='pr')
        pc = Pin(num=1, name='pc')
        r1 = Pin(num=1, name='r1')
        r2 = Pin(num=1, name='r2')
        c1 = Pin(num=1, name='c1')
        c2 = Pin(num=1, name='c2')
        d1 = Pin(num=1, name='d1')
        d2 = Pin(num=1, name='d2')
        self.pins = [pr, pc, r1, r2, c1, c2, d1, d2]
```

Is this the right approach?  I'm already getting errors trying to instantiate it (needs "fields") and I know I'll have to define the footprint somehow.

devbisme, 2021-10-04:
"How do I ...?" questions usually get more exposure if they're posted in the [discussions](https://github.com/xesscorp/skidl/discussions).

As an example, I created a SKiDL module for a PMOD plug [here](https://devbisme.github.io/circuitsascode/api/html/_modules/circuitsascode/interfaces/pmod.html#pmod_plug_12).

You could also do it like this:
```py
class mx(Part):
    def __init__(self, *args, **kwargs):
        Part.__init__(self, name="MX_SWITCH", tool=SKIDL, ref_prefix="SW", footprint="<something>", **kwargs)
        self += Pin(num=1, name='pr')
        self += Pin(num=2, name='pc')
        ...
```
After defining the `mx` class, you can instantiate it like so:
```
mx_a = mx()
```

Is there a reason why each of your pins was assigned pin number 1?

r4dr3fr4d, 2021-10-04:
'pin number 1'....Hmm not sure how I determined that...I took it out.

Do you have an example of a Python defined footprint?  Or is that still
external and we're still tied to importing existing footprints from
another source?  Are the footprints organic to kicad, in general?

Thanks for the reply - if I can get this working I'll definitely share in
the discussions and elsewhere!

devbisme, 2021-10-07:
SKiDL doesn't have any features for defining footprints. Right now, all footprints are fetched from KiCad.

The `pcbnew` package that's associated with KiCad's PCBNEW application can be used to create footprints. I've used it to make a plugin: [xess_fp_wizard](https://github.com/xesscorp/xess_fp_wizard). You could do something similar to make a Python library to make footprints.