devbisme/skidl #144: [SKiDL BUG]Paralel wires get overlayed

legalmachinery, 2021-12-19:
Hi, paralel wires in my scheme appear to be overlayed as per the attached scheme. Is there any way to get them drawn separately please. Thank you.

OS Debian 10, Python 3.7, skidl latest.
[image]

devbisme, 2021-12-20:
This is to be expected since the routing done by `generate_schematic` is very rudimentary. I'm working through @shanemmattner's code and extending it (in the `gensch` branch), but I don't expect to have a fix for problems like this in the near future. All you can do right now is manually rearrange the wiring in EESCHEMA.

s39674, 2022-02-27:
@legalmachinery can you please explain how did you even get to this point? (installation steps, how did you get that ic to show up) i can't even get to outputting any schematic. All i get is this error:

```
Traceback (most recent call last):
  File "/home/serveruser/Documents/skidl-gensch/skidl/skidlbaseobj.py", line 43, in __getattr__
    return self.__getattribute__("fields")[key]
KeyError: 'x'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/serveruser/Documents/skidl-gensch/skidlTestTree.py", line 57, in <module>
    generate_schematic()
  File "/home/serveruser/Documents/skidl-gensch/skidl/circuit.py", line 965, in generate_schematic
    gen_func(**kwargs)
  File "/home/serveruser/Documents/skidl-gensch/skidl/tools/kicad/gen_schematic.py", line 1150, in gen_schematic
    preprocess_parts_and_nets(circuit)
  File "/home/serveruser/Documents/skidl-gensch/skidl/tools/kicad/gen_schematic.py", line 160, in preprocess_parts_and_nets
    pin.pt = Point(pin.x, pin.y)
  File "/home/serveruser/Documents/skidl-gensch/skidl/skidlbaseobj.py", line 45, in __getattr__
    raise AttributeError
AttributeError
```
Is it trying to access an attribute named: "x" in the pin class? i search through that class and i can't seem to find any variable with that name.

should i put this on a separate bug report? thanks!

OS Debian , Python 3.9, skidl-gensch latest