devbisme/skidl #151: Suggesting that search() will return an object

s39674, 2022-06-13:
**Is your feature request related to a problem? Please describe.**
My project gets a text from some place and sends it into skidl `search()` function. But `search()` doesn't return an object like show() does and thus cannot be analyzed and worked with.

**Describe the solution you'd like**
I would like to see `search()` return an object.

**Additional context**
A quick demo:
```python
>>> a = show('Relay_SolidState', 'MOC3061M')
>>> a

   MOC3061M  () :
           Pin None/1/~/PASSIVE
           Pin None/2/~/PASSIVE
           Pin None/3/NC/NO-CONNECT
           Pin None/4/~/PASSIVE
           Pin None/5/NC/NO-CONNECT
           Pin None/6/~/PASSIVE
>>> type(a)
<class 'skidl.part.Part'>

# a is an object and can be work with

>>> b = search('MOC3061')
Relay_SolidState.kicad_sym: MOC3061M ( )
>>> type(b)
<class 'NoneType'>

# While b isn't
```

@devbisme Thank you for this amazing project!

devbisme, 2022-06-14:
I created a function `get_parts(terms)` which will search for parts that match the given search terms and return a list of `Part` objects. You can install it like so:
```
pip install git+https://github.com/devbisme/skidl@s39674
```
Give it a try and see if it does what you want.

s39674, 2022-06-15:
@devbisme thank you, but unfortunately I cant test this as it gives me this error:
```bash
Building wheels for collected packages: skidl
  Building wheel for skidl (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for skidl (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [2183 lines of output]
      running bdist_wheel
      The [wheel] section is deprecated. Use [bdist_wheel] instead.
      running build
      running build_py
      creating build
      creating build/lib
      creating build/lib/skidl
      copying skidl/utilities.py -> build/lib/skidl
      copying skidl/interface.py -> build/lib/skidl

...

copying skidl/tools/spice/__init__.py -> build/lib/skidl/tools/spice
      running install
      Traceback (most recent call last):
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 261, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 230, in build_wheel
          return self._build_with_temp_dir(['bdist_wheel'], '.whl',
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 215, in _build_with_temp_dir
          self.run_setup()
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 267, in run_setup
          super(_BuildMetaLegacyBackend,
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 158, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 42, in <module>
          setup(
        File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 153, in setup
          return distutils.core.setup(**attrs)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/core.py", line 148, in setup
          return run_commands(dist)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/core.py", line 163, in run_commands
          dist.run_commands()
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 967, in run_commands
          self.run_command(cmd)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 335, in run
          self.run_command('install')
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 985, in run_command
          cmd_obj.ensure_finalized()
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 107, in ensure_finalized
          self.finalize_options()
        File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 45, in finalize_options
          orig.install.finalize_options(self)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 320, in finalize_options
          self.finalize_unix()
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 59, in finalize_unix
          super().finalize_unix()
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 488, in finalize_unix
          self.select_scheme("posix_prefix")
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 55, in select_scheme
          super().select_scheme(name)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 522, in select_scheme
          scheme = _load_schemes()[name]
[... 11 lines cut ...]
          scheme['headers'] = orig_install._load_schemes()['posix_prefix']['headers']
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 137, in wrapped_load_schemes
          _inject_headers(name, scheme)
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 125, in _inject_headers
          scheme['headers'] = orig_install._load_schemes()['posix_prefix']['headers']
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 137, in wrapped_load_schemes
          _inject_headers(name, scheme)
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 125, in _inject_headers
          scheme['headers'] = orig_install._load_schemes()['posix_prefix']['headers']
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 137, in wrapped_load_schemes
          _inject_headers(name, scheme)
...

File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 103, in _load_schemes
          sysconfig_schemes = _load_sysconfig_schemes() or {}
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 91, in _load_sysconfig_schemes
          with contextlib.suppress(AttributeError):
      RecursionError: maximum recursion depth exceeded
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for skidl
Failed to build skidl
ERROR: Could not build wheels for skidl, which is required to install pyproject.toml-based projects
```

devbisme, 2022-06-15:
Not sure what is wrong on your end. I just did an install in a virtualenv using Python 3.10.4 and got this:

```bash
~/tmp $ pip install git+https://github.com/devbisme/skidl@s39674
Collecting git+https://github.com/devbisme/skidl@s39674
  Cloning https://github.com/devbisme/skidl (to revision s39674) to /tmp/pip-req-build-ppx2oib_
  Running command git clone --filter=blob:none --quiet https://github.com/devbisme/skidl /tmp/pip-req-build-ppx2oib_
  Running command git checkout -b s39674 --track origin/s39674
  Switched to a new branch 's39674'
  Branch 's39674' set up to track remote branch 's39674' from 'origin'.
  Resolved https://github.com/devbisme/skidl to commit 219a8bb2789cbe340e9e1a496af4d7558931bf17
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting graphviz
  Using cached graphviz-0.20-py3-none-any.whl (46 kB)
Collecting kinparse>=0.1.0
  Using cached kinparse-1.1.0.tar.gz (239 kB)
  Preparing metadata (setup.py) ... done
Collecting kinet2pcb
  Using cached kinet2pcb-1.0.0.tar.gz (185 kB)
  Preparing metadata (setup.py) ... done
Collecting future>=0.15.0
  Using cached future-0.18.2.tar.gz (829 kB)
  Preparing metadata (setup.py) ... done
Collecting pyparsing>=2.1.1
  Using cached pyparsing-3.0.9-py3-none-any.whl (98 kB)
Collecting hierplace
  Using cached hierplace-1.0.0.tar.gz (313 kB)
  Preparing metadata (setup.py) ... done
Using legacy 'setup.py install' for future, since package 'wheel' is not installed.
Using legacy 'setup.py install' for kinparse, since package 'wheel' is not installed.
Using legacy 'setup.py install' for kinet2pcb, since package 'wheel' is not installed.
Using legacy 'setup.py install' for hierplace, since package 'wheel' is not installed.
Building wheels for collected packages: skidl
  Building wheel for skidl (pyproject.toml) ... done
  Created wheel for skidl: filename=skidl-1.2.0-py2.py3-none-any.whl size=717832 sha256=7b36266f07785b23dadc08e5b5d30e6eded8ab7bbe677810fa6049a6838cba32
  Stored in directory: /tmp/pip-ephem-wheel-cache-l60cojw4/wheels/4c/6e/97/8a8e7a8d69075e9c5a722b31e40e218114aa4cfef838c88d2d
Successfully built skidl
Installing collected packages: hierplace, pyparsing, graphviz, future, kinparse, kinet2pcb, skidl
  Running setup.py install for hierplace ... done
  Running setup.py install for future ... done
  Running setup.py install for kinparse ... done
  Running setup.py install for kinet2pcb ... done
Successfully installed future-0.18.2 graphviz-0.20 hierplace-1.0.0 kinet2pcb-1.0.0 kinparse-1.1.0 pyparsing-3.0.9 skidl-1.2.0
WARNING: You are using pip version 22.0.4; however, version 22.1.2 is available.
You should consider upgrading via the '/home/devb/.pyenv/versions/3.10.4/envs/pytoml/bin/python3.10 -m pip install --upgrade pip' command.                  

~/tmp $ pip list                                                                                                                                   
Package    Version
---------- -------
future     0.18.2
graphviz   0.20
hierplace  1.0.0
kinet2pcb  1.0.0
kinparse   1.1.0
pip        22.0.4
pyparsing  3.0.9
setuptools 58.1.0
skidl      1.2.0
WARNING: You are using pip version 22.0.4; however, version 22.1.2 is available.
You should consider upgrading via the '/home/devb/.pyenv/versions/3.10.4/envs/pytoml/bin/python3.10 -m pip install --upgrade pip' command.  
```

s39674, 2022-06-16:
@devbisme managed to workaround it.. 
Thank you, I actually need the library name, so `options=["parts_only"]` is not what I am looking for.
But wouldn't be much easier to add here:
https://github.com/devbisme/skidl/blob/219a8bb2789cbe340e9e1a496af4d7558931bf17/skidl/part_query.py#L122-L124

`yield part, lib_file`? I already tested that and it works fine. I can happily open a PR if that's ok.

devbisme, 2022-06-17:
OK. I suggest renaming `get_parts` to `get_parts_libs` and return a list of namedtuples with `part` and `lib` fields.

devbisme, 2022-06-17:
I guess we should also rename the option from "parts_only".

devbisme, 2022-08-15:
This issue has lain fallow for a few months. Upon rethinking it, the simplest solution for you is to use a conditional list comprehension with the existing `search_parts_iter` function to get the `(part, lib_file)` entries as in the following example: 
```python
caps = [(p[2],p[1]) for p in search_parts_iter("capacitor") if p[0]=="PART"]
```
This seems cleaner than passing options into the `search_parts_iter` function that change its behavior.

closed 2022-11-07 by devbisme