devbisme/skidl #155: `generate_pcb()` causes `AttributeError: 'NoneType' object has no attribute 'SetParent'`

shanemmattner, 2022-07-28:
**Describe the bug**
Calling `generate_pcb()` results in:

```
/builddir/build/BUILD/kicad-6.0.6/pcbnew/action_plugin.cpp(163): assert "PgmOrNull()" failed in register_action().
Traceback (most recent call last):
  File "/home/shanemattner/Desktop/esp32/skidl_esp32_example/skidl_esp32_example.py", line 105, in <module>
    generate_pcb()
  File "/home/shanemattner/.local/lib/python3.10/site-packages/skidl/circuit.py", line 562, in generate_pcb
    gen_func(file_)  # Generate the PCB file from the netlist.
  File "/home/shanemattner/.local/lib/python3.10/site-packages/skidl/tools/kicad/kicad.py", line 1004, in gen_pcb
    kinet2pcb.kinet2pcb(self, file_)
  File "/home/shanemattner/.local/lib/python3.10/site-packages/kinet2pcb/kinet2pcb.py", line 161, in kinet2pcb
    fp.SetParent(brd)
AttributeError: 'NoneType' object has no attribute 'SetParent'
```

**To Reproduce**
Steps to reproduce the behavior:
1. Download [skidl_esp32_example.txt](https://github.com/devbisme/skidl/files/9205281/skidl_esp32_example.txt)
2. Rename it `.py`
3. Run file

**Expected behavior**
`kicad_pcb` file should be generated

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Fedora 36
 - Python 3.10.5
 - SKiDL version 1.1.0
 - KiCAD 6

**Additional context**
Add any other context about the problem here.

devbisme, 2022-07-30:
This error happens because `kinet2pcb` is not finding the requested footprint so it gets set to `None`. I fixed it so an error is logged and `kinet2pcb` tries to proceed even with the missing footprint. Use `pip install git+https://github.com/devbisme/kinet2pcb` to try the fix.

I also noticed your example code referenced some pins on the `CP2104` chip using names like `~{RTS}` and `~{DTR}`. Pins with these names don't exist, but SKiDL wasn't raising any errors. I fixed that in SKiDL's `development` branch so make sure to install that.

shanemmattner, 2022-07-30:
The `kinet2pcb` fix worked but it looks like `generate_pcb()` can't find footprints:

```
[shanemattner@fedora-1 skidl_esp32_example]$ python skidl_esp32_example.py
/builddir/build/BUILD/kicad-6.0.6/pcbnew/action_plugin.cpp(163): assert "PgmOrNull()" failed in register_action().
Unable to find footprint ESP32-WROOM-32 in ${KICAD6_FOOTPRINT_DIR}/RF_Module.pretty
Unable to find footprint C_0603_1608Metric in ${KICAD6_FOOTPRINT_DIR}/Capacitor_SMD.pretty
Unable to find footprint C_0603_1608Metric in ${KICAD6_FOOTPRINT_DIR}/Capacitor_SMD.pretty
Unable to find footprint QFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm in ${KICAD6_FOOTPRINT_DIR}/Package_DFN_QFN.pretty
Unable to find footprint C_0603_1608Metric in ${KICAD6_FOOTPRINT_DIR}/Capacitor_SMD.pretty
Unable to find footprint C_0603_1608Metric in ${KICAD6_FOOTPRINT_DIR}/Capacitor_SMD.pretty
Unable to find footprint C_0603_1608Metric in ${KICAD6_FOOTPRINT_DIR}/Capacitor_SMD.pretty
Unable to find footprint USB_A_Molex_67643_Horizontal in ${KICAD6_FOOTPRINT_DIR}/Connector_USB.pretty
Unable to find footprint R_0603_1608Metric in ${KICAD6_FOOTPRINT_DIR}/Resistor_SMD.pretty
Unable to find footprint R_0603_1608Metric in ${KICAD6_FOOTPRINT_DIR}/Resistor_SMD.pretty

No errors or warnings found while creating PCB.
```
But the footprints are in the folder and KiCAD has no problem finding them:
```
[shanemattner@fedora-1 footprints]$ ls RF_Module.pretty | grep ESP32
ESP32-S2-MINI-1.kicad_mod
ESP32-S2-MINI-1U.kicad_mod
ESP32-S2-WROVER.kicad_mod
ESP32-S3-WROOM-1.kicad_mod
ESP32-S3-WROOM-1U.kicad_mod
ESP32-WROOM-32.kicad_mod
ESP32-WROOM-32U.kicad_mod
[shanemattner@fedora-1 footprints]$ ls RF_Module.pretty | grep ESP32-WROOM-32
ESP32-WROOM-32.kicad_mod
ESP32-WROOM-32U.kicad_mod
[shanemattner@fedora-1 footprints]$ ls Capacitor_SMD.pretty/ | grep C_0603_1608Metric
C_0603_1608Metric.kicad_mod
C_0603_1608Metric_Pad1.08x0.95mm_HandSolder.kicad_mod
[shanemattner@fedora-1 footprints]$ ls Package_DFN_QFN.pretty/ | grep QFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm
HVQFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm.kicad_mod
HVQFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm_ThermalVias.kicad_mod
QFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm.kicad_mod
QFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm_ThermalVias.kicad_mod
WQFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm.kicad_mod
WQFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm_ThermalVias.kicad_mod
```
[image]

Lastly, I think those names for `cp2104` pins are correct.  Here's a print out of all the pin names:
```
>>> for i in cp2104.pins:
...     print(i.name)
... 
~{RI}
NC
GPIO.3
RS485/GPIO.2
RXT/GPIO.1
TXT/GPIO.0
~{SUSPEND}
VPP
SUSPEND
~{CTS}
~{RTS}
GND
RXD
TXD
~{DSR}
~{DTR}
~{DCD}
GND
D+
D-
VIO
VDD
REGIN
VBUS
~{RST}
```

devbisme, 2022-07-30:
`kinet2pcb` does environment variable expansion. Are you sure the `KICAD6_FOOTPRINT_DIR` variable is set in the shell where you're running the script?

My CP2104 looks like this. It must be KiCad V5:
```python
CP2104 (CP2104): Single-Chip USB-to-UART Bridge, USB 2.0 Full-Speed, 2Mbps UART, QFN package
    Pin U1/1/RI,p1,RI/INPUT
    Pin U1/10/VIO/NC,VIO/NC,p10/POWER-IN
    Pin U1/11/GPIO.3,GPIO.3,p11/BIDIRECTIONAL
    Pin U1/12/GPIO.2,p12,GPIO.2/BIDIRECTIONAL
    Pin U1/13/GPIO.1,p13,GPIO.1/BIDIRECTIONAL
    Pin U1/14/GPIO.0,p14,GPIO.0/BIDIRECTIONAL
    Pin U1/15/~SUSPEND,p15,~SUSPEND/OUTPUT
    Pin U1/16/VPP,p16,VPP/PASSIVE
    Pin U1/17/SUSPEND,SUSPEND,p17/OUTPUT
    Pin U1/18/CTS,CTS,p18/INPUT
    Pin U1/19/RTS,p19,RTS/OUTPUT
    Pin U1/2/GND,GND,p2/POWER-IN
    Pin U1/20/RXD,p20,RXD/INPUT
    Pin U1/21/TXD,TXD,p21/OUTPUT
    Pin U1/22/DSR,DSR,p22/INPUT
    Pin U1/23/DTR,DTR,p23/OUTPUT
    Pin U1/24/DCD,DCD,p24/INPUT
    Pin U1/25/PAD,PAD,p25/POWER-IN
    Pin U1/3/D+,p3,D+/BIDIRECTIONAL
    Pin U1/4/D-,D-,p4/BIDIRECTIONAL
    Pin U1/5/VIO,p5,VIO/POWER-IN
    Pin U1/6/VDD,p6,VDD/POWER-IN
    Pin U1/7/REGIN,REGIN,p7/POWER-IN
    Pin U1/8/VBUS,p8,VBUS/INPUT
    Pin U1/9/~RST,~RST,p9/BIDIRECTIONAL
```

shanemmattner, 2022-08-01:
I added `KICAD6_FOOTPRINT_DIR` to my `.bashrc` file and `generate_pcb()` is working!  Interestingly, there's still an error when I try to use `digikey-footprints`:

```
[shanemattner@fedora-1 skidl_esp32_example]$ python skidl_esp32_example.py
/builddir/build/BUILD/kicad-6.0.6/pcbnew/action_plugin.cpp(163): assert "PgmOrNull()" failed in register_action().
Traceback (most recent call last):
  File "/home/shanemattner/Desktop/esp32/skidl_esp32_example/skidl_esp32_example.py", line 134, in <module>
    generate_pcb()
  File "/home/shanemattner/.local/lib/python3.10/site-packages/skidl/circuit.py", line 562, in generate_pcb
    gen_func(file_)  # Generate the PCB file from the netlist.
  File "/home/shanemattner/.local/lib/python3.10/site-packages/skidl/tools/kicad/kicad.py", line 1004, in gen_pcb
    kinet2pcb.kinet2pcb(self, file_)
  File "/home/shanemattner/.local/lib/python3.10/site-packages/kinet2pcb/kinet2pcb.py", line 158, in kinet2pcb
    lib_uri = fp_libs[fp_lib]
KeyError: 'digikey-footprints'
```
And I made sure it's in the directory with all the same permissions other libraries have:
```
[shanemattner@fedora-1 skidl_esp32_example]$ ll $KICAD6_FOOTPRINT_DIR | grep digikey-footprints
drwxr-xr-x.   1 root root  28K Jul 31 21:09 digikey-footprints.pretty
drwxr-xr-x. 1 root root  28K Jul 31 21:12 digikey-footprints.pretty
```

devbisme, 2022-08-01:
What is happening is that your `digitkey-footprints.pretty` directory doesn't have an entry in the `fp-lib-table` file that KiCad and `kinet2pcb` uses to find footprint libraries. Go looking for that file and add the `digikey-footprints.pretty` directory using the same format as the others. Then it should work.

devbisme, 2022-08-02:
To make things like using the DigiKey footprints easier, I added an `fp_libs` option to `generate_pcb()` so you can pass a list of footprint directories.  For example, I put my DigiKey footprints in the directory `~/tmp/digikey-footprints.pretty` so I modified the `generate_pcb()` call to:
```python
generate_pcb(fp_libs=["~/tmp/digikey-footprints.pretty"])
```
You'll need to install the `development` branch of `skidl` and the `master` branch of `kinet2pcb` to use this feature.

devbisme, 2022-08-15:
Is there any more to do on this?

shanemmattner, 2022-08-15:
All good.  Thank you!

closed 2022-08-15 by shanemmattner