devbisme/skidl #158: Circuit simulation error: cannot load library 'libngspice.so'

shanemmattner, 2022-08-05:
Running the first SKiDL simulation example `sim = circ.simulator()` results in the error:
```
Exception has occurred: OSError
cannot load library 'libngspice.so': libngspice.so: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngspice.so'
  File "/home/shanemattner/Desktop/skidl_spice_test/sandbox.py", line 24, in <module>
    sim = circ.simulator()                 # Create a simulator for the Circuit object.
```

**To Reproduce**
Steps to reproduce the behavior:
1. run the 'current through resistor' example in SKiDL documentation

**Expected behavior**
Simulation should run like example

**Screenshots**
I followed [this](https://github.com/PySpice-org/PySpice/issues/117) thread about the problem and confirmed that `ngspice` is installed
```
from skidl import *
print(lib_search_paths)

# Load the SKiDL + PySpice packages and initialize them for doing circuit simulations.
from skidl.pyspice import *
print(lib_search_paths)

from ctypes.util import find_library

ngspice_lib_filename = find_library('ngspice')
print(ngspice_lib_filename)
```

Results in:

```
{'kicad': ['.', '/usr/share/kicad/symbols'], 'skidl': ['.', '/home/shanemattner/.local/lib/python3.10/site-packages/skidl/libs'], 'spice': ['.']}
{'kicad': ['.', '/usr/share/kicad/symbols'], 'skidl': ['.', '/home/shanemattner/.local/lib/python3.10/site-packages/skidl/libs'], 'spice': ['.']}
libngspice.so.0
```
**Desktop (please complete the following information):**
 - Fedora 36
 - Python 3.10
 - SKiDL version 

**Additional context**
Add any other context about the problem here.

devbisme, 2022-08-06:
Have you tried creating some links to the shared libraries as in this [thread](https://github.com/PySpice-org/PySpice/issues/234)?

shanemmattner, 2022-08-07:
After creating the appropriate symbolic link I was able to run the example until [this](https://github.com/devbisme/skidl/issues/159) issue blocked the file from running.

Here's the command for anyone running Fedora 36
```
[shanemattner@fedora-1 lib64]$ sudo ln -s libngspice.so.0.0.4 libngspice.so
[shanemattner@fedora-1 lib64]$ ls -l | grep ngspice
lrwxrwxrwx. 1 root root        19 Aug  6 18:21 libngspice.so -> libngspice.so.0.0.4
lrwxrwxrwx. 1 root root        19 May 20 00:31 libngspice.so.0 -> libngspice.so.0.0.4
-rwxr-xr-x. 1 root root  10464656 May 20 00:45 libngspice.so.0.0.4
drwxr-xr-x. 1 root root       120 Aug  4 19:48 ngspice
```

closed 2022-08-07 by shanemmattner