devbisme/skidl #159: module 'skidl.pyspice' has no attribute 'node'

shanemmattner, 2022-08-07:
**Describe the bug**
Trying to run the first example from [pyspice](https://github.com/devbisme/skidl/blob/master/examples/spice-sim-intro/spice-sim-intro.ipynb) and can't find `node`

Core issue may be:
```Unsupported Ngspice version 37```
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Copy paste `current through resistor` code from [here](https://github.com/devbisme/skidl/blob/master/examples/spice-sim-intro/spice-sim-intro.ipynb)
2. Run code

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Fedora 36
 - Python 3.10
 - SKiDL 1.1.0

**Additional context**
Add any other context about the problem here.

devbisme, 2022-08-07:
I'm really not sure what's happening here. The code isn't exactly like what you linked to because it uses `sp.node()` instead of just `node()`. I assume you did something like `import skidl.pyspice as sp`. I can't see why that would cause a problem, but who knows? Is this some problem caused by running the notebook cell out of order with the preceding cells? Have you tried restarting the kernel and running from the start?

You can also insert something like `print(dir(sp))` before the line that causes the problem and see what attributes are present in `sp`. Maybe `sp` isn't what you think it is.

E-wallace0001, 2022-09-19:
I did a straight copy and paste in to both a python terminal, and Jupyter Notebook to no joy.
I am getting :
` NameError: name 'node' is not defined`
I can find the definition in tools>spice.
and I can see that spice is being pulled in from pyspice.py with the import
`from .tools import SKIDL, SPICE`
I initially tried it with a straight pip pull, and then installed it from this git repo to the same error.
From the manual install, it flagged up a syntax error.
`  File "build/bdist.linux-x86_64/egg/skidl/libs/skidl_lib_sklib.py", line 6
    Part(**{ 'name':'EXPI', 'dest':TEMPLATE, 'tool':SKIDL, 'fall_time_constant':5, 'rise_delay_time':5, 'initial_value':5, 'pyspice':{'name': 'ExponentialCurrentSource', 'kw': {'initial_value': 'initial_value', 'pulsed_value': 'pulsed_value', 'rise_delay_time': 'rise_delay_time', 'rise_time_constant': 'rise_time_constant', 'fall_delay_time': 'fall_delay_time', 'fall_time_constant': 'fall_time_constant', 'p': 'node_plus', 'n': 'node_minus'}, 'add': <function add_part_to_circuit at 0x7fd01478aee0>}, '_aliases':Alias({'EXPONENTIALCURRENT', 'exponentialcurrent', 'expi'}), 'rise_time_constant':5, 'description':'Exponential current source', 'fall_delay_time':5, 'pulsed_value':5, 'keywords':'exponential current source', 'match_pin_substring':False, 'ref_prefix':'I', 'num_units':1, 'fplist':None, 'do_erc':True, 'aliases':Alias({'EXPONENTIALCURRENT', 'exponentialcurrent', 'expi'}), 'pin':None, 'footprint':None, 'pins':[
                                                                                                                                                                                                                                                                                                                                                                                                                                                                    ^
SyntaxError: invalid syntax
`
For ease, this is pointing to:
`'add': <function add_part_to_circuit at 0x7fd01478aee0>`
Of which, I've never seen this kind of code being used in python and don't like the look of; I replaced it with the function name it's trying to call rather than a memory address as I believe the intention is just a pointer to the function to be called, this resolves the install error, but not the missing function.

This is on Ubuntu 20.04 with python 3.8.10.

E-wallace0001, 2022-09-20:
I was able to resolve by changing the import in libs/pyspice_sklib.py
from 
`from skidl.tools.spice import (
    add_part_to_circuit,
    add_xspice_to_circuit,
    not_implemented,
)`
to 
`from skidl.tools.spice import *`

E-wallace0001, 2022-09-20:
@devbisme I will try to do a push later; I seem to be having an issue with permissions to do a push to skidl repo

devbisme, 2022-12-05:
I just checked the `development` branch and commit `0add7d3` on 11/07/2021 contains similar code to what is in the PR from @PolarState. I think this problem might disappear if you use the `development` branch instead of the version 1.1.0 because that was released on 10/23/2021. At least this is what I found with my testing.

PolarState, 2022-12-08:
Thanks Dave!

devbisme, 2022-12-08:
You're welcome, although I must apologize that there hasn't been a new version of SKiDL on PyPi for over a year so you had to fix a problem that was already solved but hadn't been released. But I am planning a new release soon with new features including schematic generation (thanks, @shanemmattner!).