devbisme/skidl #163: WARNING: Could not load KiCad schematic library "Device", falling back to backup library.

QGB, 2022-11-05:
```
WARNING: Could not load KiCad schematic library "Device", falling back to backup library. @ [C:\test\jlc-eda\<frozen importlib._bootstrap>:228=>C:\QGB\babun\cygwin\bin\qgb\tests\skidl_test.py:20]
11-05 14:00:50 skidl 118:Could not load KiCad schematic library "Device", falling back to backup library. @ [C:\test\jlc-eda\<frozen importlib._bootstrap>:228=>C:\QGB\babun\cygwin\bin\qgb\tests\skidl_test.py:20]
WARNING: Could not load KiCad schematic library "Device", falling back to backup library. @ [C:\test\jlc-eda\<frozen importlib._bootstrap>:228=>C:\QGB\babun\cygwin\bin\qgb\tests\skidl_test.py:21]
11-05 14:00:50 skidl 118:Could not load KiCad schematic library "Device", falling back to backup library. @ [C:\test\jlc-eda\<frozen importlib._bootstrap>:228=>C:\QGB\babun\cygwin\bin\qgb\tests\skidl_test.py:21]
WARNING: Could not load KiCad schematic library "power", falling back to backup library. @ [C:\test\jlc-eda\<frozen importlib._bootstrap>:228=>C:\QGB\babun\cygwin\bin\qgb\tests\skidl_test.py:28]
11-05 14:00:50 skidl 118:Could not load KiCad schematic library "power", falling back to backup library. @ [C:\test\jlc-eda\<frozen importlib._bootstrap>:228=>C:\QGB\babun\cygwin\bin\qgb\tests\skidl_test.py:28]
WARNING: Could not load KiCad schematic library "power", falling back to backup library. @ [C:\test\jlc-eda\<frozen importlib._bootstrap>:228=>C:\QGB\babun\cygwin\bin\qgb\tests\skidl_test.py:29]
11-05 14:00:50 skidl 118:Could not load KiCad schematic library "power", falling back to backup library. @ [C:\test\jlc-eda\<frozen importlib._bootstrap>:228=>C:\QGB\babun\cygwin\bin\qgb\tests\skidl_test.py:29]

No errors or warnings found while generating netlist.

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\QGB/babun/cygwin/bin\qgb\tests\skidl_test.py", line 44, in <module>
    print(generate_pcb())
  File "C:\Users\qgb\Documents\KiCad\6.0\3rdparty\Python39\site-packages\skidl\circuit.py", line 562, in generate_pcb
    gen_func(file_)  # Generate the PCB file from the netlist.
  File "C:\Users\qgb\Documents\KiCad\6.0\3rdparty\Python39\site-packages\skidl\tools\kicad\kicad.py", line 1004, in gen_pcb
    kinet2pcb.kinet2pcb(self, file_)
  File "C:\Users\qgb\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kinet2pcb\kinet2pcb.py", line 219, in kinet2pcb
    fp_lib, fp_name = part.footprint.split(":")
AttributeError: 'NoneType' object has no attribute 'split'
```

devbisme, 2022-11-07:
You haven't provided the source for skidl_test.py. I can't provide any help without that. Sorry.

QGB, 2022-11-12:
https://github.com/QGB/QPSU/blob/3219c7f09412c5364b9ca3839100eb6100ea2710/tests/skidl_test.py

devbisme, 2022-11-14:
The AttributeError is raised because you have parts which do not have any assigned footprints. Since there footprint is None, the split() function is raising an exception. There should have been some error reports about the parts not having footprints, so I'm not sure why those weren't generated.

The solution to this is to define some footprints for the parts you are using.