devbisme/skidl #165: [SKiDL BUG] Brackets not handled in component name

Timot05, 2022-11-12:
`my_led = Part(lib="default_lib.kicad_sym", name="0805G_(Green)", dest=TEMPLATE)`

returns:

`ValueError: Unable to find part 0805G_(Green) in library default_lib.kicad_sym.`

Skidl does not seems to be able to find this component. Is that on purpose or should we fix it?

devbisme, 2022-11-14:
I assume you're using the SKiDL 1.1.0.

I created the same error by adding parentheses to a part name in a KiCad V6 library. I think the parentheses are getting interpreted as part of the nested S-expression parentheses used in V6 symbol libraries. I'll have to investigate and make some changes in the V6 library handler. For now, you'll just have to use names that don't include parentheses.

Thanks for reporting this error!

Timot05, 2022-11-17:
Got it! I'll stick to names without parenthesis until then.