devbisme/skidl #166: [SKiDL BUG] generate_svg() consistently raises UnboundLocalError

mawildoer, 2022-11-14:
**Describe the bug**
I've run `generate_svg()` on a couple of scripts now that successfully `generate_netlist`s, but it consistently raises an exception `UnboundLocalError: local variable 'num_units' referenced before assignment` (with a traceback)

**To Reproduce**
Steps to reproduce the behavior:
1. Install skidl 1.1.0 (latest as of writing, but the exception occurs in 1.0.0 too)
2. Add `generate_svg()` to the bottom of a seemingly otherwise functional script (see snippet below)
3. Execute script

Snippet:
```python
from skidl import *

# Create input & output voltages and ground reference.
vin, vout, gnd = Net('VI'), Net('VO'), Net('GND')

# Create two resistors.
r1, r2 = 2 * Part("Device.kicad_sym", 'R', TEMPLATE, footprint='Resistor_SMD.pretty:R_0805_2012Metric')
r1.value = '1K'   # Set upper resistor value.
r2.value = '500'  # Set lower resistor value.

# Connect the nets and resistors.
vin += r1[1]      # Connect the input to the upper resistor.
gnd += r2[2]      # Connect the lower resistor to ground.
vout += r1[2], r2[1] # Output comes from the connection of the two resistors.

generate_svg()
```

**Expected behavior**
An SVG to be generated, per docs here: https://devbisme.github.io/skidl/#svg-schematics

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Mac OSx 12
 - Python version: Python 3.10.6
 - SKiDL version 1.1.0

**Additional context**
As I mentioned at the top, I've attempted this on a few other designs, but it's consistently arising. I feel like this is something likely to have been raised before me though, so not sure what's up. Thank you for your assistance. It's greatly appreciated!

devbisme, 2022-11-14:
Thanks for the clear error report! I've replicated the error. I'll try to push a correction to the development branch today.

devbisme, 2022-11-14:
I've opened an enhancement issue and closed this one.

closed 2022-11-14 by devbisme

mimidbe, 2023-05-20:
Good morning,
the problem still persists!
I installed skidl **version 1.1.0** , and use it under **jupyter**.
  and I get the following error:

[image]

for the _same previous file_!

Thank you for your assistance.